ddvk/remarkable2-recovery #1: What to do after you have the devices mounted?

aribo, 2020-10-31:
After hunting for the parts, tinkering with them until I could get a reasonable stable pogo (well rather "bare pin") connection, now I go four mounted devices: Home, Boot, RootA and RootB. I think the device boots from Root A by default, because this where the /etc/systemd/system/multi-user.target.wants/xochitl.service is missing, in Room B is still there. However, I cannot work with the drives, it doesn't let me upload anything to them. Any ideas?

ddvk, 2020-10-31:
just use sudo cp

Eeems, 2020-10-31:
> just use sudo cp

Shouldn't it be a link?

aribo, 2020-10-31:
Yes, i managed now to  copy the xochitl.service, but funny enough now the device doesn't want to shut off (?)

aribo, 2020-10-31:
i actually had to copy the xochitl.service file from rootB copy it intp lib/systemd/system and then ln -s to /etc/systemd/system/multi-user.target.wants/
All this with sudo. It worked, but now the device doesn't turned off. When I push the button for 10 secs it flips and back to "is starting" screen.

aribo, 2020-10-31:
Ok, now it worked! I did the following:
- sudo cp /rootB/lib/system/systemd/xochitl.service /rootA/lib/system/systemd/xochitl.service
- sudo cp /rootA/lib/system/systemd/xochitl.service /rootA/etc/systemd/system/xochitl.service
- sudo ln  /rootA/lib/system/systemd/xochitl.service /rootA/etc/systemd/system/multi-user.target.wants/xochitl.service

Thank you very much @ddvk for all your help. You've been very patient.

ddvk, 2020-10-31:
> > just use sudo cp
> 
> 
> 
> Shouldn't it be a link?

yes, i meant he should copy the link  from the other root

closed 2021-02-11 by ddvk