ddvk/remarkable2-recovery #2: Questions about the hardware

ahmedshah1494, 2020-11-19:
I have a couple of questions about the setup. 

1-The usb jack attached to the pogo pins is a micro-usb jack, right?

2-Is the jack on the usb-C breakout a female? Do you then use a usb-C to usb-C to connect the breakout board to the usb-C port on the remarkable?

Thanks

ddvk, 2020-11-19:
1. yes, usb micro
2. i use male, but there are people that have used a female one with a
cable (i didnt have a usb 3.1 cabel) as long as the all the pins are
connected

ahmedshah1494, 2020-11-19:
Thanks a lot for responding.
Just to make sure I got this right, I need to get a male usb-C breakout with a B8 pin, plug it into the powered off device, and attach a 10K resistor between GND and B8, right?

ddvk, 2020-11-19:
yes, then power up

starnix59, 2021-01-05:
Where did you get the pogo adapter?

ddvk, 2021-01-05:
there are links in the readme to aliexpress

isman7, 2021-01-06:
Hi! I'm buying also the components to have access to recovery, OTG, etc (https://github.com/dps/remarkable-keywriter/issues/14). Just found these kind of pogo connectors with magnets: https://es.aliexpress.com/item/1005001587511989.html. For anyone interested.

starnix59, 2021-01-08:
In case this helps anyone that is impatient and doesn't want to wait for their AliExpress order to come in and you own or have access to a 3D printer. :smile: 

Here is a support bracket to use breadboard patch cables as "pogo" workarounds. It's a bit touchy, but it will work in a pinch.

https://www.thingiverse.com/thing:4714318

isman7, 2021-02-01:
Hey, for anyone interested, as it has been discussed in https://github.com/dps/remarkable-keywriter/issues/14, pogo pins with magnets do not work directly on the remarkable 2, because magnets are not aligned. However, you can use a female connector for the magnets and the male connector for the actual connection. And with some soldering do the trick. Pics:

https://user-images.githubusercontent.com/9478220/106371365-cf8d7b00-6363-11eb-99ec-afc09332f91f.png
https://user-images.githubusercontent.com/9478220/106392383-c945e000-63f1-11eb-9fcb-1b5599f292d6.png
https://user-images.githubusercontent.com/9478220/106390405-9d722c80-63e8-11eb-9007-86e1df8c19b7.gif

alistair23, 2021-02-02:
That's great! How reliable/easy to re-connect is it?

isman7, 2021-02-02:
The idea is to use the magnets, so it's magnetic, the problem is that we have only magnets in one side of the connector, so the Pogo pins have one degree of liberty that we don't want. But, connecting it in plain surface i.e. the folio, suffices to have a good connection.

What I'm thinking  is to explore an integration with the USB-C connector all in one board or something, this would fix the free liberty degree.

I've made this gif to clarify the re-connection:

[image]

Hope it resolves some doubts!

alistair23, 2021-02-04:
That is awesome! So that is just two male pogo adapters?

ddvk, 2021-02-04:
i made a holder in freecad, but havent ironed its kinks yet, i ll push the file

isman7, 2021-02-04:
Thanks! Well, as they are sailed in pairs, I've used the male adapter as
the actual connector and the female adapter for the magnets. Possible
improvements are:

- Add an LED to check power as a hint of the connection.
- New version with a pre-layout strip PCB board to reduce soldering.
- New version with actual magnets instead of the female connector, this
would add some stability, using 4 magnets, instead of 2. This will reduce
the liberty of the mount. To do so I need some 3D printing as well, thus I
need some time to play with the design.

I really didn't found any docs about the magnets so, some days ago I played
with magnets they are composite magnets with reversed polarity `+-+-`, I
did some review on Twitter (It's in Catalan, so you can google translate or
enjoy the pictures!):
https://twitter.com/ismansiete/status/1352681340777463810

alistair23, 2021-02-08:
Any update on the holder @ddvk ?

ddvk, 2021-02-10:
> Any update on the holder @ddvk ?
it is not yet optimal, but i pushed the version I have

PepikVaio, 2021-03-07:
> there are links in the readme to aliexpress
> […](#)

Hi, this link is probably bad.
https://www.aliexpress.com/i/4000026719770.html
Can I use this? Or another?
https://www.aliexpress.com/item/4000221550930.html

ddvk, 2021-03-07:
> > there are links in the readme to aliexpress
> 
> > […](#)
> 
> 
> 
> Hi, this link is probably bad.
> 
> https://www.aliexpress.com/i/4000026719770.html
> 
> Can I use this? Or another?
> 
> https://www.aliexpress.com/item/4000221550930.html

this one is female, you'll need a cable that has a b8 connection

PepikVaio, 2021-03-07:
Ok, I can't find the right one :(

PepikVaio, 2021-03-08:
> > > there are links in the readme to aliexpress
> > 
> > 
> > > […](#)
> > 
> > 
> > Hi, this link is probably bad.
> > https://www.aliexpress.com/i/4000026719770.html
> > Can I use this? Or another?
> > https://www.aliexpress.com/item/4000221550930.html
> 
> this one is female, you'll need a cable that has a b8 connection

OK, I found this: It will be work?
Please tell me, if I will be need some next parts?
Thank you.
https://usb.brando.com/type-c-test-male-pcb-board-with-pin-header_p19685c0067d015.html
http://www.snailshop.cz/soucastky-do-nepajiveho-pole/1935-usb-micro-b-connector-breakout-board.html
http://www.snailshop.cz/vodice-pro-nepajive-pole/2699-vodice-dutinka-dutinka-10cm-40ks.html
https://www.gme.cz/rm-10k-0207-0-6w-1

Eeems, 2025-07-21:
As there has been no activity on this in several years, I'm closing the issue.

closed 2025-07-21 by Eeems