ddvk/remarkable2-recovery #4: Clarification on Notes

AqsaKashaf, 2020-12-07:
Hi, 

I could not understand what you mean by these:
  -> B8 has to be connected to the same ground as the Pogo GND pin
  -> The ID pin on Pogo connector is not connected to anything

In the pictures you have connected the ID pin to the connector. And i don't understant how to connect B8 and pogo GND pin.

ddvk, 2020-12-07:
that was added by someone else because they couldn't get it to work. I use B1 for GND

AqsaKashaf, 2020-12-11:
Hi Thank you for the reply, I have tried many different things, I think my connection is stable enough, but every time I get the message in dmesg that "Cannot enable. Maybe the usb cable is bad? attempt power cycle" I have tried different cables. To debug this, I need some information. Can you please tell me why are we pulling down b8? Also, do you have any idea what might be wrong

starnix59, 2021-01-07:
I am having the same issue as @AqsaKashaf "Cannot enable. Maybe the usb cable is bad? attempt power cycle".

Any recommendations? I have tried multiple usb cables and usb ports, but no luck yet.

boxfire, 2021-01-07:
I found the GPIO is only on one side, i.e. A8 and B8 are not both connected. Practically this means just flip over your breakout or use A8. Don't get the POGO pins swapped, you can overload your host device.

ddvk, 2021-01-07:
i thought it was self evident, but i ll add a clarification that orientation matters

starnix59, 2021-01-07:
I am having intermittent issues with getting the error "Cannot enable. Maybe the USB cable is bad? attempt power cycle" or nothing at all in dmesg. I looked at the connections and they appear to be solid. I am waiting on the pogo part and was trying to use some headers and a 3d printed support. 

I feel like I am missing something simple, any ideas where to look?

ddvk, 2021-01-07:
it is mostly the cable, but make sure you didnt reverse d+ and d-

starnix59, 2021-01-07:
I double-checked and the pinout appears to be correct (GND,ID,D+,D-,VBUS). Just in case my cable color coding was incorrect or the breakout board for USB mini I swapped the D+ & D- I started getting the following. It doesn't appear to be consistent, after I restarted the tablet I no longer see those messages.

new full-speed USB device number 26 using xhci_hcd
Device not responding to setup address.
Device not responding to setup address.
device not accepting address 27, error -71
unable to enumerate USB device

AqsaKashaf, 2021-01-07:
In my case, the usb-c breakout board was faulty. I changed the board and it started working

starnix59, 2021-01-07:
Well, some good news, and a thank you to @AqsaKashaf my usb-c breakout board was an issue. After swapping that I am able to get consistent results. 

Currently, I am getting these errors which are slightly different from some of the prior ones.

new full-speed USB device number 102 using xhci_hcd
device descriptor read/64, error -71
device descriptor read/64, error -71

attempt power cycle

new full-speed USB device number 103 using xhci_hcd
Device not responding to setup address.
Device not responding to setup address.
device not accepting address 103, error -71
unable to enumerate USB device

starnix59, 2021-01-07:
Good news, I got past the issues and recovered the ssh password. In case anyone else runs into this issue, charge the battery for a couple of hours (apparently the battery drains pretty quick when it's stuck). 

One issue that I am running into is xochitl is not running on boot even though the service is enabled. If I ssh in and start the service manually it works though until you reboot. Any ideas on that one?

Eeems, 2021-01-07:
@starnix59 Anything in the logs? `journalctl -u xochitl`

starnix59, 2021-01-07:
@Eeems nothing in the logs

-- Logs begin at Tue 2021-01-05 16:58:01 UTC, end at Thu 2021-01-07 00:09:27 UTC
. --
-- No entries --

Eeems, 2021-01-08:
Are you sure it's enabled?

starnix59, 2021-01-08:
This is the output of systemctl list-unit-files --type=service

[[0;4mUNIT FILE                              STATE          [[0m
autovt@.service                        [[0;1;32menabled        [[0m
busybox-ifplugd@.service               indirect
busybox-udhcpd@.service                [[0;1;31mdisabled       [[0m
console-getty.service                  [[0;1;31mdisabled       [[0m
container-getty@.service               static
crashuploader.service                  [[0;1;32menabled        [[0m
dbus-org.freedesktop.hostname1.service static
dbus-org.freedesktop.locale1.service   static
dbus-org.freedesktop.login1.service    static
dbus-org.freedesktop.network1.service  [[0;1;32menabled        [[0m
dbus-org.freedesktop.timedate1.service static
dbus.service                           static
debug-shell.service                    [[0;1;31mdisabled       [[0m
dhcpcd.service                         [[0;1;32menabled        [[0m
disable-wowlan.service                 [[0;1;32menabled        [[0m
dropbear@.service                      static
dropbearkey.service                    static
emergency.service                      static
getty@.service                         [[0;1;32menabled        [[0m
halt-local.service                     static
initrd-cleanup.service                 static
initrd-parse-etc.service               static
initrd-switch-root.service             static
initrd-udevadm-cleanup-db.service      static
kmod-static-nodes.service              static
qmdns-responder.service                [[0;1;32menabled        [[0m
rc-local.service                       static
remarkable-fail.service                static
remarkable-reboot.service              [[0;1;32menabled        [[0m
remarkable-shutdown.service            [[0;1;32menabled        [[0m
rescue.service                         static
run-postinsts.service                  [[0;1;32menabled        [[0m
serial-getty@.service                  indirect
system-update-cleanup.service          static
systemd-ask-password-console.service   static
systemd-ask-password-wall.service      static
systemd-exit.service                   static
systemd-firstboot.service              static
systemd-fsck-root.service              enabled-runtime
systemd-fsck@.service                  static
systemd-halt.service                   static
systemd-hibernate-resume@.service      static
systemd-hibernate.service              static
systemd-hostnamed.service              static
systemd-hwdb-update.service            static
systemd-hybrid-sleep.service           static
systemd-initctl.service                static
systemd-journal-catalog-update.service static
systemd-journal-flush.service          static
systemd-journald.service               static
systemd-kexec.service                  static
systemd-localed.service                static
systemd-logind.service                 static
systemd-machine-id-commit.service      static
systemd-modules-load.service           static
systemd-networkd-wait-online.service   [[0;1;32menabled        [[0m
systemd-networkd.service               [[0;1;32menabled        [[0m
systemd-poweroff.service               static
systemd-random-seed.service            static
systemd-reboot.service                 static
systemd-remount-fs.service             static
systemd-rfkill.service                 static
systemd-suspend.service                static
systemd-sysctl.service                 static
systemd-sysusers.service               static
systemd-timedated.service              static
systemd-timesyncd.service              [[0;1;32menabled        [[0m
systemd-tmpfiles-clean.service         static
systemd-tmpfiles-setup-dev.service     static
systemd-tmpfiles-setup.service         static
systemd-udev-settle.service            static
systemd-udev-trigger.service           static
systemd-udevd.service                  static
systemd-update-done.service            static
systemd-update-utmp-runlevel.service   static
systemd-update-utmp.service            static
systemd-vconsole-setup.service         static
systemd-volatile-root.service          static
tarnish.service                        [[0;1;32menabled        [[0m
update-engine.service                  [[0;1;32menabled        [[0m
user@.service                          static
var-volatile-cache.service             [[0;1;32menabled        [[0m
var-volatile-lib.service               [[0;1;32menabled        [[0m
var-volatile-spool.service             [[0;1;32menabled        [[0m
var-volatile-srv.service               [[0;1;32menabled        [[0m
wpa_supplicant-nl80211@.service        [[0;1;31mdisabled       [[0m
wpa_supplicant-wired@.service          [[0;1;31mdisabled       [[0m
wpa_supplicant.service                 [[0;1;31mdisabled       [[0m
wpa_supplicant@.service                [[0;1;31mdisabled       [[0m
xochitl.service                        [[0;1;32menabled        [[0m

90 unit files listed.

Eeems, 2021-01-08:
What does `systemctl status xochitl` give you?

starnix59, 2021-01-08:
systemctl status xochitl returns

xochitl.service - reMarkable main application
   Loaded: loaded (/lib/systemd/system/xochitl.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

If I manually start the service with systemctl start xochitl it works until reboot

Eeems, 2021-01-08:
That's very odd. I'd suggest taking a look at the full `journalctl -b` output to see if it sheds any light on what's happening.

starnix59, 2021-01-08:
Thank you @Eeems that pointed to the problem, tarnish.service was trying to start but no longer existed. 

Jan 05 16:58:17 reMarkable systemd[256]: [[0;1;31m[[0;1;39m[[0;1;31mtarnish.service: Failed to execute command: No such file or directory[[0m
Jan 05 16:58:17 reMarkable systemd[256]: [[0;1;31m[[0;1;39m[[0;1;31mtarnish.service: Failed at step EXEC spawning /opt/bin/tarnish: No such file or directory[[0m
Jan 05 16:58:17 reMarkable systemd[1]: [[0;1;39m[[0;1;31m[[0;1;39mtarnish.service: Main process exited, code=exited, status=203/EXEC[[0m
Jan 05 16:58:17 reMarkable systemd[1]: [[0;1;39m[[0;1;31m[[0;1;39mtarnish.service: Failed with result 'exit-code'.[[0m
Jan 05 16:58:22 reMarkable systemd[1]: tarnish.service: Service hold-off time over, scheduling restart.
Jan 05 16:58:22 reMarkable systemd[1]: tarnish.service: Scheduled restart job, restart counter is at 4.
Jan 05 16:58:22 reMarkable systemd[1]: Stopped oxide system service.
Jan 05 16:58:22 reMarkable systemd[1]: Started oxide system service.

nyc-tinker, 2021-01-30:
Hi @starnix59 - how did you resolve the issue below?  Sounds like it resolved.  I tried like 4 different cables, and i'm always getting the message below.  Did it work from replacing the USB-C breakout board?  Or some other fix?  

> new full-speed USB device number 102 using xhci_hcd
> device descriptor read/64, error -71
> device descriptor read/64, error -71
>
> attempt power cycle
>
> new full-speed USB device number 103 using [...]

alistair23, 2021-01-30:
@nyc-tinker You seem to have a different issue? Is that the log from your host machine? In which case your connection is not secure. You need to ensure all the pins are connected and remain connected.

NOTE: Generally after I see those messages I reboot the rM2, otherwise both my host machine and rM2 get in a strange state and I have trouble getting a USB connection again.

nyc-tinker, 2021-01-30:
Yes, these are from the host: and they are largely ￼identical to what was posted above; the RM2 is powered off when I'm attaching the pogo pins to the device. These error messages on the host keep cycling for the duration that the pogo pins are attached to the device￼. ￼I am wondering whether this kind of error message is specific enough to isolate to a specific hardware issue, such as a bad USB-C breakout board or a bad USB Type B board, or a faulty cable￼.

￼So you're saying to power on the device (unplug the USB C breakout board)￼, then restart, then power off again?  Considering the device is off for all this I guess it's not obvious to me why restarting the device would do anything. ￼￼ I will give it a shot anyway and report back. ￼

> @nyc-tinker You seem to have a different issue? Is that the log from your host machine? In which case your connection is not secure. You need to ensure all the pins are connected and remain [...]

alistair23, 2021-01-30:
Your rM2 will turn on when plugged in, so it's probably powered on.

It's not specific enough to know where the problem is. Just that somewhere there is an electrical problem.

You don't need to unplug anything. I hold the power button for 16 seconds to turn it off then hold the button again for 3 seconds to turn it on. That won't fix the problem with the bad connection, but if you have wiggled things to try and fix it I usually have to reboot to get it to work again.

nyc-tinker, 2021-01-31:
Resolved by re-soldering the USB B breakout board (connected to the pogo pins).  So now it recognizes the device.  Electrical connection was the culprit.

However, imx_usb, when run, indicates no job found.  I tried the arch linux AUR package  imx_usb and also tried a compiling the binary, and both result in the same outcome.

Output is below.  I feel like it is tantalizingly close to working!

```
config file <.//imx_usb.conf>
vid=0x066f pid=0x3780 file_name=mx23_usb_work.conf
vid=0x15a2 pid=0x004f file_name=mx28_usb_work.conf
vid=0x15a2 pid=0x0052 file_name=mx50_usb_work.conf
vid=0x15a2 pid=0x0054 file_name=mx6_usb_work.conf
vid=0x15a2 pid=0x0061 file_name=mx6_usb_work.conf
vid=0x15a2 pid=0x0063 file_name=mx6_usb_work.conf
vid=0x15a2 pid=0x0071 file_name=mx6_usb_work.conf
vid=0x15a2 pid=0x007d file_name=mx6_usb_work.conf
vid=0x15a2 pid=0x0080 file_name=mx6ull_usb_work.conf
vid=0x1fc9 pid=0x0128 file_name=mx6_usb_work.conf
vid=0x15a2 pid=0x0076 file_name=mx7_usb_work.conf
vid=0x1fc9 pid=0x0126 file_name=mx7ulp_usb_work.conf
vid=0x15a2 pid=0x0041 file_name=mx51_usb_work.conf
vid=0x15a2 pid=0x004e file_name=mx53_usb_work.conf
vid=0x15a2 pid=0x006a file_name=vybrid_usb_work.conf
vid=0x066f pid=0x37ff file_name=linux_gadget.conf
vid=0x1b67 pid=0x4fff file_name=mx6_usb_sdp_spl.conf
vid=0x0525 pid=0xb4a4 file_name=mx6_usb_sdp_spl.conf
vid=0x1fc9 pid=0x012b file_name=mx8mq_usb_work.conf
vid=0x1fc9 pid=0x0134 file_name=mx8mm_usb_work.conf
vid=0x1fc9 pid=0x013e file_name=mx8mn_usb_work.conf
vid=0x3016 pid=0x1001 file_name=mx8m_usb_sdp_spl.conf
config file <.//mx7_usb_work.conf>
parse .//mx7_usb_work.conf
no job found

```

> Your rM2 will turn on when plugged in, so it's probably powered on.
> 
> It's not specific enough to know where the problem is. Just that somewhere there is an electrical problem.
> 
> You don't need to unplug anything (assuming the resistor is still connected). I hold the power button for 16 seconds to turn it off then hold the button again for 3 seconds to turn it on. That won't fix the problem with the bad connection, but if you have wiggled things to try and fix it I usually have to reboot to get it to work again.

alistair23, 2021-01-31:
That probably means the rM2 isn't connected or maybe you should run the imx program as root.

nyc-tinker, 2021-01-31:
Success!  What wasn't clear (although in retrospect should have been obvious) is that the config files in this repo are needed to direct imx_usb to the hardware; not the [main repo](https://github.com/boundarydevices/imx_usb_loader) for imx_usb.  

Executing `sudo ./imx_usb` in the cloned repo (`git clone https://github.com/ddvk/remarkable2-recovery.git`), I was able to find the block devices.

Thanks @alistair23 for the help!

ddvk, 2021-02-11:
silved then

closed 2021-02-11 by ddvk