ddvk/remarkable2-recovery #7: Linux Developement

alistair23, 2021-01-13:
The [u-boot source](https://github.com/reMarkable/uboot/blob/zero-sugar/include/configs/zero-sugar.h#L144) indicates that u-boot will first try to boot a kernel/initrd/dtb from memory before falling back to loading a kernel from the rootFS.

Has anyone been able to flash images to boot a kernel from USB?

I'm guessing something along the lines of:

```shell
./imx_usb zImage -l0x82000000 zero-sugar.dtb -l0x88000000 rootfs.cpio -l0x83800000
```

It also [seems like rM1](https://github.com/ddvk/remarkable-uuuflash#linux) has serial access, any such luck on rM2?

alistair23, 2021-01-13:
Good news on the UART, it looks like [UART6 in the device tree](https://github.com/reMarkable/linux/blob/zero-sugar/arch/arm/boot/dts/zero-sugar.dts#L620) is exposed via the pogo pins with a pin mux: https://github.com/reMarkable/linux/blob/zero-sugar/drivers/misc/rm-otgcontrol/otgcontrol_onewire.c#L132

Just need to figure out how to enable it.

ddvk, 2021-01-13:
i think uart6 is just for the otg mode switching (1-wire stuff).

Looking at the imx7drm it seems usbotg1_oc is multiplexed with uart3, but not much.  
usb_otg_dp and _dn are not muxed.

alistair23, 2021-01-14:
Progress!

From Linux I can write serial out over the ID pin on the OTG Pogo pins. It just takes a few changes to the device tree.

I have also changed the Linux console to print all boot output to the ID pin. I can see a handful of kernel log messages but not all of them. I'm not sure why I can't see all of them yet.

alistair23, 2021-01-14:
I have now also started to look at booting via the pogo pins. My `mx7_usb_work.conf` now has this

```
mx7
hid,1024,0x910000,0x80000000,1G,0x00900000,0x20000
zImage:load 0x80800000
zero-sugar.dtb:load 0x83000000
```
and I can successfully download the images but I can't get them to boot. Any ideas what I am missing? Do I need to load u-boot as well with `u-boot.imx:clear_dcd,plug,jump header2`?

alistair23, 2021-01-15:
Tried

```
zImage:load 0x80800000
zero-sugar.dtb:load 0x83000000
rootfs.cpio:load 0x83800000
u-boot-dtb.imx:clear_dcd,plug,jump header
```

and no luck. Seems like it might be loading something, but unable to see any UART output

alistair23, 2021-01-17:
Closing this. I have my own u-boot, Linux and OE rootFS booting with UART out over the pogo pins. Still to UART Rx though, but I haven't really tried to see if it works.

I also now have the latest 5.11-rc3 kernel running on the reMarkable 2 (without full support though). I have started sending patches to mainline Linux.

closed 2021-01-17 by alistair23

Eeems, 2021-01-18:
> Closing this. I have my own u-boot, Linux and OE rootFS booting with UART out over the pogo pins. Still to UART Rx though, but I haven't really tried to see if it works.
>
> I also now have the latest [...]

Any chance you could open a PR adding some of your learnings to the REAME?

alistair23, 2021-01-18:
I have started adding some bits to the wiki: https://remarkablewiki.com/tech/rm2_otg_pogo

I'm happy to write something up though. Any details on what specifically you would like?

Eeems, 2021-01-18:
Perhaps just a link to the wiki page will be enough?

alistair23, 2021-01-18:
It's not super helpful, there are still a few patches and config changes required.

I have a working OE rootFS. I'm planning on getting that up and running fully, at that point I'll add a REAME and send a PR to meta-freescale-3rdparty. I'll add all the documentation there.

ddvk, 2021-01-18:
did you modify u-boot to set the pin,  you could probably get the serial output

alistair23, 2021-01-19:
Yep, I have u-boot working as well.

I just submitted a PR for this repo with some details and some more pictures: https://github.com/ddvk/remarkable2-recovery/pull/8

Full software support is available here: https://github.com/Freescale/meta-freescale-3rdparty/pull/116

Next step is upstreaming, if anyone wants to help let me know. My current tree is here: https://github.com/alistair23/linux/tree/zero-sugar-mainline