ddvk/remarkable2-recovery #11: No communication on pogo pin port

pabloaul, 2021-09-11:
I'm having bad luck getting anything like "Freescale SemiConductor" to show up on dmesg via the pogo pins. 

I soldered the USB cable directly to the pogo pins on PCB and checked continuity with my multimeter and all seems fine.
Soldered a 10kOhm resistor on test point #3 (SBU2) and later just to make sure on test point #4 too (SBU1).

Deattaching the Pogo Vbus does show regular USB errors in the terminal and plugging in via USB-C also shows the ""cdc_ether"" device making me assume that the device just boots normally and terminal works as intended. Tested different USB ports on computer, no success. All done with both (un)plugged battery to make sure that it is shutdown.

Struggling with this since more than a day now because I forgot to note/change the ssh key and the GUI is not going beyond the splashscreen...

ReMarkable 2 device on the latest beta firmware 2.10.0.295

pabloaul, 2021-09-12:
Maybe an easier question to ask for now would be: Is any activity in the terminal expected when SBU is not properly bridged?

ddvk, 2021-09-16:
post photos of the setup

something is probably not connected

jmpesp, 2021-09-27:
Note SBU2 is not symmetric in the cable:

[image]

What worked for me was flipping the USB-C cable :)

pabloaul, 2022-01-05:
sadly no luck
[image]
This is how the setup currently looks like.
This time I'm using a proper breakout board, but the device still keeps booting into OS. When I have the main usb-c connected (with b8 still pulled down) I can observe it finding an ethernet adapter device which makes me assume that it is not going into recovery at all.

alistair23, 2022-01-05:
Try flipping the USB-C board. It usually only works one way around

pabloaul, 2022-01-05:
I've tried pretty much all I could do there including flipping. Starting to think this one is a lost cause...

jmpesp, 2022-01-06:
@pabloaul your setup looks much more stable than mine! I didn't solder the micro usb connector.

Note I specifically used a [USB-C 3.1 gen 2 cable][1], so that may be something to try.

[1]: https://www.amazon.ca/gp/product/B07D7SP4T4

jmpesp, 2022-01-06:
My setup was rm2's usbc connected to a [USB-C breakout board that includes SBU2, and I used a breadboard to attach the pull down resistor.](https://operand.ca/images/2021/09/rm2_recovery/recovery_hardware.jpg)

Your USB-C breakout board is connected directly so my cable suggestion may not be valid.

terev, 2022-02-25:
I'm having basically the same issue as @pabloaul . I disabled the default launcher and I guess enabling the oxide launcher failed without me noticing. Then rebooted without writing down the ssh key and now It's just stuck on the splash screen. It won't even turn off when holding power. 

I'm just getting USB enumeration errors when trying to make a connection. My last ditch effort will be to find a pogo connector and learn how to solder lol. Hoping its just an unstable connection.

rndmono, 2022-05-01:
@terev Since you are getting enumeration errors, you might want to try enforcing a USB 1.x connection, which is likely more stable and could work (it did for me). The easiest way might be to use an old USB Hub which only supports USB 1.x. Enforcing a lower speed on the host PC might be possible as well, but depends on your hardware (did not work with my PCs).

pabloaul, 2024-12-20:
After building up motivation to look at this device again under the microscope, I've noticed that I had torn two pads which was why it wasn't able to boot into recovery.
[image]
Fixing those two spots made everything work for me and now the device starts up again after fixing my mess from 3 years ago :)

I will close this issue since it is a very specific case of me being insane to open the device to bridge the SBUS2.

closed 2024-12-20 by pabloaul