ddvk/remarkable2-recovery #12: Problem with unbricking

Amatrelan, 2021-09-18:
Classical error and forgot to take ssh password up before starting to toy with rM2...

Well now when I try to unbrick with this guide for some reason I cannot make it to _USB Mass Storage device detected_ state.

I have pogo connection as described and other connections too. Tested with multimeter that resistance is correct and other things too (there was little variance on resistor batch I have, so might be that it's the problem but 10.02k...)

Well now to the logs I have:

> [ 2771.537305] hid-generic 0003:15A2:0076.0025: hiddev97,hidraw7: USB HID v1.10 Device [Freescale SemiConductor Inc  SE Blank ULT1] on usb-0000:03:00.0-7/input0
> [ 2831.535253] usb 1-7: USB disconnect, device number 69
> [ 2832.898015] usb 1-7: new high-speed USB device number 70 using xhci_hcd
> [ 2833.107290] usb 1-7: New USB device found, idVendor=15a2, idProduct=0076, bcdDevice= 0.01
> [ 2833.107294] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [ 2833.107296] usb 1-7: Product: SE Blank ULT1
> [ 2833.107297] usb 1-7: Manufacturer: Freescale SemiConductor Inc
> [ 2833.126303] hid-generic 0003:15A2:0076.0026: hiddev97,hidraw7: USB HID v1.10 Device [Freescale SemiConductor Inc  SE Blank ULT1] on usb-0000:03:00.0-7/input0

It seems to disconnect quite frequently automatically and reconnects, never found out the reason it does it, as you can see it's already on `69` and going up. Never found out reason it does this. I tried multiple different cables and micro-usb breakout board but nothing fixed it.

Also it seems to my eyes that `imx_usb` is run correctly:

> config file <.//imx_usb.conf>
> vid=0x15a2 pid=0x0076 file_name=mx7_usb_work.conf
> config file <.//mx7_usb_work.conf>
> parse .//mx7_usb_work.conf
> Trying to open device vid=0x15a2 pid=0x0076
> Interface 0 claimed
> HAB security state: development mode (0x56787856)
> == work item
> filename u-boot-ums.imx
> load_size 0 bytes
> load_addr 0x00000000
> dcd 0
> clear_dcd 0
> plug 0
> jump_mode 3
> jump_addr 0x00000000
> == end work item
> header_max=10400
> 
> loading binary file(u-boot-ums.imx) to 877effd4, skip=0, fsize=9f02c type=aa
> 
> <<<651308, 651308 bytes>>>
> succeeded (security 0x56787856, status 0x88888888)
> jumping to 0x877f08e4
 
But still it won't show as mountable device (`lsblk`). But I noticed it disconnects after `imx_usb` but it won't connect as bootable, only reconnects to "old" way.

So if someone can point out right away whats wrong, I'm happy to listen. 

also is in this stage `if the device does not start on its own, press the Power Button` meant to show up "Remarkable is booting" text on screen or just device showing up in `lsusb` / `dmesg`, if it means that remarkable text then I have some problem cause while usb-c breakout board is on it won't boot at all on that state.

ddvk, 2021-09-18:
after you see Freescale etc, you should remove the B8 short, before you run ./imx_usb

Amatrelan, 2021-09-18:
Yeah I remove it, but it still does that "hard" reboot thing after running `imx_usb`.

ddvk, 2021-09-18:
the cable pogo connection is not stable then

Amatrelan, 2021-09-19:
Ok, seems to be that in my re2 there is some problem in pogo pins, on my friend re2 same setup works perfectly and no problems at all.. well seems to be that my re2 gets upgraded from `digital paper` to `paper weight`.

closed 2021-09-19 by Amatrelan