ddvk/remarkable2-recovery #15: over-current condition

ZhigangPu, 2022-04-11:
Hi all,

I can only see "over-current condition" on my `dmesg -w` output. What's wrong?

I guess it is because I use the wrong side of pogo pin. But after a flip the pogo pin, there is no output in dmesg. However, I can see "flash" picture on the screen of RM2 near the battery icon at  the bottom left corner which means charging.

Any help will be highly appreciated!

For your better reference, here is my dmesg output
`[  865.255349] usb usb3-port1: over-current condition
[  865.471266] usb usb4-port2: over-current condition
[  865.471345] usb usb3-port2: over-current condition
[  865.686794] usb usb4-port3: over-current condition
[  865.686861] usb usb3-port3: over-current condition
[  865.903407] usb usb4-port4: over-current condition
[  865.912894] usb usb3-port4: over-current condition
[  866.126704] usb usb3-port5: over-current condition
[  866.126707] usb 3-5: USB disconnect, device number 44
[  866.502712] usb usb3-port6: over-current condition
[  866.502714] usb 3-6: USB disconnect, device number 43
[  866.870705] usb usb4-port1: over-current condition
[  866.870733] usb usb3-port1: over-current condition
[  867.087245] usb usb4-port2: over-current condition
[  867.087318] usb usb3-port2: over-current condition
[  867.302745] usb usb3-port3: over-current condition
[  867.302757] usb usb4-port3: over-current condition
[  868.006740] usb 3-5: new full-speed USB device number 45 using xhci_hcd
[  868.166710] usb usb4-port3: over-current condition
[  868.383227] usb usb4-port4: over-current condition
[  868.599216] usb usb4-port1: over-current condition
[  868.815253] usb usb4-port2: over-current condition
[  868.982700] usb usb3-port6: over-current condition
[  869.030704] usb usb4-port3: over-current condition
[  869.342658] usb usb3-port1: over-current condition
[  869.558662] usb usb3-port2: over-current condition
[  869.774656] usb usb3-port3: over-current condition
[  869.990653] usb usb3-port4: over-current condition
[  870.206653] usb usb3-port5: over-current condition
[  870.838666] usb 3-6: new high-speed USB device number 46 using xhci_hcd
[  870.989709] usb 3-6: New USB device found, idVendor=05e3, idProduct=0608, bcdDevice=88.32
[  870.989711] usb 3-6: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  870.989712] usb 3-6: Product: USB2.0 Hub
[  871.026518] hub 3-6:1.0: USB hub found
[  871.027105] hub 3-6:1.0: 4 ports detected
[  872.383159] usb 3-5: new full-speed USB device number 47 using xhci_hcd
[  872.550236] usb 3-5: New USB device found, idVendor=1462, idProduct=7c37, bcdDevice= 0.01
[  872.550238] usb 3-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  872.550240] usb 3-5: Product: MYSTIC LIGHT 
[  872.550240] usb 3-5: Manufacturer: MSI
[  872.550241] usb 3-5: SerialNumber: A02019050806
[  872.571544] input: MSI MYSTIC LIGHT  as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-5/3-5:1.0/0003:1462:7C37.000E/input/input40
[  872.571728] hid-generic 0003:1462:7C37.000E: input,hiddev0,hidraw2: USB HID v1.10 Device [MSI MYSTIC LIGHT ] on usb-0000:2a:00.3-5/input0
`

Best,
zhigang

adin666666, 2022-04-11:
Are you a chinese, if so I prefer 中文。

The remarkable 2 is used by my son as a middle school student, he is just using it as a notebook as usual. So I don’t think there is some action to pogo pin by him.

I have no idea on your dmesg as well. What I have done is having a look at its directory tree and adding chinese font ttf to font directory and a new subdirectory opentype/. That’s all we have done before.

Now we cannot SSH to the new rM2 ever with right password identified by preference by SSH from MacOS, DCommander from MacOS, SCP from windows 11. What’s wrong with it? please help us. I want to put chinese font ttf to it to let my son read chinese ePUB book on it.

It is unnecessary for us to do any more on it. I have another eBook with Andornoid and a order for a new color Andornoid eBook.

b rgds.
Hongqing

adin666666, 2022-04-11:
你好， 另外，我们自己做了一个外包装来一同和rM2一起使用。原因主要是rM2用手举着使用的时候它的重量让手指不舒服。所以，我们没有机会误触旁边的触点。
我儿子主要是用Mac mini 的USB端口来给rM2充电，供参考。

我儿子的动手能力还不足以动rM2的硬件，而我呢，已经到了不愿意动手去改动硬件的年龄了，连软件我现在都已经失去主动动手的兴趣了。

b rgds.

ddvk, 2022-04-11:
what are you trying to achieve?

ZhigangPu, 2022-04-11:
Hi ddvk，
I haven't seen "USB HID v1.10 Device [Freescale SemiConductor Inc SE Blank ULT1]" from `dmesg -w` after RM2 restarted. That bothers me.

ZhigangPu, 2022-04-11:
@[adin666666](https://github.com/adin666666) 你好，我也在解决这个ssh的问题，但是遇到了一些问题，如果我可以解决这个问题，我会分享过程给你。

adin666666, 2022-04-11:
好，谢谢。
因为rM2能够支持中文对我们来说太重要了，rM2确实是个好东西。😁

ddvk, 2022-04-11:
you have to power it down, short/pulldown B8, then press power button. and still it has to have a really good connection with the pogo pins

ZhigangPu, 2022-04-11:
Yes, I have powered RM2 down and short B8 and B12 with 10k resistor. How to make sure a really good connection? I can see a flash icon near the battery icon at bottom left corner of RM2, I think connection is good. But `dmesg -w` output contains no information about new device.

tomcattiger1230, 2022-04-23:
@ZhigangPu 

Hi, I have tried the provided method. It works indeed, but not so easy. I had a comparable situation. But in my case, it is because of the connection quality of the USB cable. If you have followed the introduction, you should see a new USB showing in dmesg. Several points you may try: 1. change the side of the USB-C connector; 2. check you connection with the USB cable. Furthermore, my RM2 was not totally started during the procedure, it was holding on to the starting screen, but the USB device was shown in my system. The flash icon should not be shown even if you have a good connection (since in my test the system was not totally started). Indeed, it may indicate that you have a bad procedure (connect -> disconnect (system totally starts)-> connect).

Eeems, 2023-04-21:
> You might have had the pogo pin in wrong direction. I got this same "over-current" message when the "v-bus" was on the far side of usbc. However for me, I could find anything in dmesg even if the pogo ping was in the right direction.

From https://github.com/ddvk/remarkable2-recovery#guide

> **Make sure the orientation is correct, you risk frying your device if you connect the pins in the wrong order**

You may have fried the USB stack in the device, so it will no longer work if you have it connected in reverse.

wordpr3ss, 2023-04-21:
... how do I know for sure it's fried?

I always made sure it's in the right direction before, but no luck. so I tried the reverse direction in desperation...

Eeems, 2023-04-21:
> ... how do I know for sure it's fried?
> 
> I always made sure it's in the right direction before, but no luck. so I tried the reverse direction in desperation...

If it isn't showing up, and all the other variables are good (good contact, it has enough charge, you are properly pulling down B8, USB cable and port are good etc), it's quite likely it's fried.

Eeems, 2025-07-21:
As there has been no follow-up in several years, I'm closing the issue.

closed 2025-07-21 by Eeems