ddvk/remarkable2-recovery #17: how to mount partition and get root password

magnum, 2022-11-12:
after 
```
sudo ./imx_usb
```

i receive this output
```
config file <.//imx_usb.conf>
vid=0x15a2 pid=0x0076 file_name=mx7_usb_work.conf
config file <.//mx7_usb_work.conf>
parse .//mx7_usb_work.conf
Trying to open device vid=0x15a2 pid=0x0076
Interface 0 claimed
HAB security state: development mode (0x56787856)
== work item
filename u-boot-ums.imx
load_size 0 bytes
load_addr 0x00000000
dcd 0
clear_dcd 0
plug 0
jump_mode 3
jump_addr 0x00000000
== end work item
header_max=10400

loading binary file(u-boot-ums.imx) to 877effd4, skip=0, fsize=9f02c type=aa

<<<651308, 651308 bytes>>>
succeeded (security 0x56787856, status 0x88888888)
jumping to 0x877f08e4
```

what do I have to to to mount the correct partition and see the password?

Eeems, 2022-11-12:
https://github.com/ddvk/remarkable2-recovery#next-steps

As for mounting the partition, you should see a new mountable block device that you can mount with the following commands:
```bash
mkdir tmpmount
mount /dev/<block-device> tmpmount/
```
Make sure to replace `<block-device>` with the new block device.

You can then run the following to get the password, and then unmount it:
```bash
grep Developer tmpmount/root/.config/remarkable/xochitl.conf
```

magnum, 2022-11-14:
how can i find the <block-device> ? 
from the imx_usb output?

Eeems, 2022-11-14:
Check the `dmesg` output and look in `/dev` for new block devices

magnum, 2022-11-15:
i'm going mad with that pogo pins, it's almost impossible to understand if they're connecting wrongly or the device is not the  current state.

anyway, this is my output, here's what a i get when (1 tiem in 100) i get a useful response from in dmesg stream
[image]

most of the times i get nothing or messages like these
[image]

Eeems, 2022-11-15:
I don't see `USB Mass Storage device detected` like the readme mentions.

magnum, 2022-11-15:
@Eeems i've never seen "USB Mass Storage device detected" is it a problem? 

i was thinking that seeing this code (after many tries moving an retrying with pogo pins)
```
[ 5316.786251] us 1-1.1: Product: SE Blank ULT1
[ 5316.786268] us 1-1.1: Manufacturer: Freescale SemiConductor Inc
[ 5316.790414] hid-generic 0003:15A2:0076.006E: hiddev99, hidraw: USB HID v1.10 Device [Freescale SemiConductor Inc SE Blank ULT1] on usb-0000:01:00.0-1.1/ input0
```
was the sign that the tablet was in recovery mode, the connection via the pogo pins was succesful and a device was found: isn't it?

Eeems, 2022-11-15:
The next step is to remove the pulldown resistor, then run the imx_usb executable, and then you would see that message.

magnum, 2022-11-15:
good luck to me to reach that point again :-) 
i'm still fighting to understand if the tablet is in recovery mode (or not)

just for confirmation, i'm using this breakout board
[image]

when you say b8 and b12, b12 is gnd? 
is it connected correctly?  considering the tablet's screen, in my case pins should be up or down?
in a photo linked in the readme, they seem to connect b8 and gnd, and between them there are 4 free pins,
in my case if i connect b8 and b12 (gnd) there are 3 free pins between them !?!?

anyway, thank you so much for your time, patience and support

ddvk, 2022-11-15:
mostly likely unstable pogo connection

magnum, 2022-11-20:
i managed to see `SE Blank ULT1`
but how can i found the block device name? 

i've listed /dev before and after the conection,
i've found a new `/dev/usb/hiddev1` but then i try to mount it with 
```
sudo mount /dev/usb/hiddev1 tmpmount/ 
```
it says
```
/dev/usb/hiddev1 tmpmount/  is not a block device
```

i've got a lot of dirs under `/dev/block`, something like...
```
259:0 259:1 259:2 259:3 259:4 7:0 7:1 7:2 7:3 7:4 7:5 7:6 7:7 7:8 7:9 8:0 8:1 8:2 8:3 8:4
```
how can i find the block device if under that `/dev/block` dir ?

this is the ouput ouf sudo `sudo ./imx_usb`

```
config file <.//imx_usb.conf>
vid=0x15a2 pid=0x0076 file_name=mx7_usb_work.conf
config file <.//mx7_usb_work.conf>
parse . / /mx7_usb_work.conf
Trying to open device vid=0x15a2 pid=0x0076
Interface 0 claimed
HAB security state: development mode (0x56787856)
== work item
filename U-boot-uns.imx
load_size o bytes 
load _addr 0x00000000
dcd 0 
clear dcd 0
plug 0
jump_mode 3
jump addr 0x00000000
== end work item 
header max=10400

loading binary file(u-boot-uns. imx) to 877effd4, skip-0, fsize-9f02c type=aa

<<<651308, 651308 bytes>>>
succeeded (security 0x56787856, status 0X88888888)
jumping to 0x877f08e4
```

Eeems, 2022-11-20:
What does dmesg say after you run imx_usb? It should tell you what new devices there are. Another thing to confirm is that you have good contact with the pogo connector.

magnum, 2022-11-20:
this is the output dmesg when i connect the pogo pins
```
[ 3735.863163] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3735.863166] usb 1-1: Product: SE Blank ULT1
[ 3735.863169] usb 1-1: Manufacturer: Freescale Semiconductor Inc
[ 3735.865226] hid-generic 0003:15A2:0076.004D: hiddevo, hidraw1: USB HID v1.10 D evice [Freescale SemiConductor Inc SE Blank ULT1] on usb-0000:00:14.0-1/input0
```

after i run `sudo ./imx_usb` then dmesg output adds these lines
```
[ 4081.739047] usb 1-1: USB disconnect, device number
[ 4082.049528] usb 1-1: new high-speed USB device number 48 using xhci_hed
[ 4082.198919] usb 1-1: New USB device found, idVendor-15a2, idProduct-0076, bcd Device= 0.01
[ 4082.198927] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 4082.198930] usb 1-1: Product: SE Blank ULT1
[ 4082.198932] usb 1-1: Manufacturer: Freescale Semiconductor Inc
[ 4082.200888] hid-generic 0003:15A2:0076.0056: hiddevo,hidraw1: USB HID V1.10 Device Freescale SemiConductor Inc SE Blank ULT1] on usb-0000:00:14.0-1/input0
```

Eeems, 2022-11-20:
I still don't see _USB Mass Storage device detected_ like in the readme. To confirm, you are removing the pulldown resistor before running imx_usb as per the readme right?

https://github.com/ddvk/remarkable2-recovery#guide

magnum, 2022-11-20:
ok now i've done it and a few usb devices have been auto-mounted by ubuntu
i tried to do this
```
sudo mount /dev/sdb4 tmpmount/ 
```
cannot find on 
```
tmpmount/root/.config 
```
but only
```
tmpmount/root/.entware
```
that's empty...

Eeems, 2022-11-20:
That is the only folder/file on the partition?

magnum, 2022-11-20:
yep

magnum, 2022-11-20:
i read https://toltec-dev.org/
and i've done a 
```
sudo rm <don't remember which usb mounted device>/etc/systemd/system/xochitl.service.d/toltec-wrapper.conf
```
but the tablet stil doesn't restart....it show "is starting"

Eeems, 2022-11-20:
Did you have toltec installed? If your data partition was empty, it sounds like you tried to do a factory reset. You'll need to make sure to run the changes to remove toltec-wrapper.conf from the correct root partition, so I'd recommend both.

magnum, 2022-11-20:
yes i did a factory reset
but it didn't restarted after this
shame on me, i forgot to remove toltec BEFORE doing a factory reset
now, how can solve this? 

referring to 
https://toltec-dev.org/
i was able to to remove only the startup toltec file
```
rm /etc/systemd/system/xochitl.service.d/toltec-wrapper.conf
```
but obviously not to execute this commands
```
systemctl daemon-reload
systemctl reset-failed xochitl
systemctl restart xochitl
```

i launched weeks ago, and fter that all stoped working.
i started all this recover procedure, only to be able to recover the root password and execute those commands to remove toltec, but now the partition is empty so i'm able to get a root password? 
so how can i have the tablet boot again, even with a reset, without toltec, i just need it to be operation with the stock os version?

Eeems, 2022-11-20:
https://remarkablewiki.com/trouble/factory-reset

Since you did a factory reset, we can't get the root password from the `xochitl.conf` file, since it no longer exists.

If you remove `toltec-wrapper.conf` from both root partitions, there is no need to run the other commands, since that's just for a running system.

Are there any other files in `/etc/systemd/system/xochitl.service.d/`?

magnum, 2022-11-20:
so i need to connect again and search for 2 root partitions and remove all the ```toltec-wrapper.conf```and maybe the ```xochitl.service.d``` directories?

Eeems, 2022-11-20:
Correct. You can also use [journalctl](https://www.man7.org/linux/man-pages/man1/journalctl.1.html) to read the logs from the `/var/log/journal` directory to see what happened on the last boot to see if there is any further information on what is going wrong. For example: `journalctl --directory tmpmount/var/log/journal -eu xochitl`

https://www.howtogeek.com/499623/how-to-use-journalctl-to-read-linux-system-logs/

magnum, 2022-11-21:
@Eeems i'm on a google meet rigth now :-) if you have few minutes, we'll speed up the things and iìll be geteful for life :-) 
https://meet.google.com/jbg-uxwb-pmx

Eeems, 2022-11-21:
I don't use google meet. If you need more instant help, I'm sure there are people in the community discord willing to help.

magnum, 2022-11-21:
i'm running ubuntu live from a usb dongle
there a the partitions mounted
```
Filesystem                   Size  Used Avail Use% Mounted on
tmpfs                        765M   19M  746M   3% /run
/dev/sda1                    3.6G  3.6G     0 100% /cdrom
/cow                         3.8G  244M  3.5G   7% /
/dev/disk/by-label/writable   55G  1.4M   52G   1% /var/log
tmpfs                        3.8G     0  3.8G   0% /dev/shm
tmpfs                        5.0M  8.0K  5.0M   1% /run/lock
tmpfs                        3.8G  720K  3.8G   1% /tmp
tmpfs                        765M  180K  764M   1% /run/user/999
/dev/sdb4                    6.6G   32K  6.2G   1% /home/ubuntu/Desktop/remarkable2-recovery/tmpmout
/dev/sdc2                    232M  213M  2.4M  99% /media/ubuntu/239f74ba-a120-4bf4-a650-259daff8ea98
/dev/sdc4                    6.6G   32K  6.2G   1% /media/ubuntu/home
/dev/sdc1                     20M  172K   20M   1% /media/ubuntu/BOOT
/dev/sdc3                    240M  202M   22M  91% /media/ubuntu/43a6b54e-abb4-4c79-8191-176c0097b983
```

in which partitions i've to search for `/etc/systemd/system/xochitl.service.d/toltec-wrapper.conf` ? i've found only one and deleted it

these are the journalctl logs from both
journalctl --directory /media/ubuntu/239f74ba-a120-4bf4-a650-259daff8ea98/var/log/journal/ > ../journalctl_log1.txt
journalctl --directory /media/ubuntu/43a6b54e-abb4-4c79-8191-176c0097b983/var/log/journal/ > ../journalctl_log2.txt

[journalctl_log1.txt](https://github.com/ddvk/remarkable2-recovery/files/10051194/journalctl_log1.txt)

[journalctl_log2.txt](https://github.com/ddvk/remarkable2-recovery/files/10051198/journalctl_log2.txt)

Eeems, 2022-11-21:
Both of those logs are not recent. The first one is from October 27th, the second from March 25th. I'd assume the first is the most recent and the one being used?
When you had toltec installed, were you using a launcher like Oxide or draft? If so, you'll have to manually re-enable xochitl by adding the systemd unit file link manually to the disk.

magnum, 2022-11-21:
> When you had toltec installed, were you using a launcher like Oxide or draft? 
yes i remember i test something like that

how can i manually re-enable xochitl by adding the systemd unit file link manually to the disk. ?

Eeems, 2022-11-21:
https://unix.stackexchange.com/a/159293

magnum, 2022-11-21:
ok what's the soruce path for the symlink?

magnum, 2022-11-21:
and after all i can try to connect using
```
ssh -oHostKeyAlgorithms=+ssh-rsa -v root@10.11.99.1
```
but what's the password !?!?
is it written to any file i can read?

Eeems, 2022-11-21:
You need a `xochitl.conf` to be generated to know what the password is, which means you need xochitl to be running.

`/lib/systemd/system/xochitl.service` is the location of the service file.

magnum, 2022-11-21:
so i'll need something like

```
ln -s /usr/lib/systemd/system/netctl.service \
      /etc/systemd/system/multi-user.target.wants/netctl.service
```

Eeems, 2022-11-21:
Right, where the second path starts with tmpmount. So probably this: `ln -s /lib/systemd/system/xochitl.service tmpmount/etc/systemd/system/multi-user.target.wants/xochitl.service`

magnum, 2022-11-21:
i'm not using tmpmount because ubuntu automounts under `/media/something.... `the 2 partitions
o you think the symlinks will still work?

magnum, 2022-11-21:
it's present both on
```
etc/systemd/system/xochitl.service.d/xochitl.service
etc/systemd/system/multi-user.target.wants/xochitl.service
```
but it still doesn't start
i'm wasting a lot of time to keep pogo pin connection up...

Eeems, 2022-11-21:
`sudo ln -s /lib/systemd/system/xochitl.service etc/systemd/system/xochitl.service.d/xochitl.service`
The first path must be valid when in the booted system.

Eeems, 2022-11-21:
There are two root partitions, the active one, and the fallback. When you upgrade, the fallback is overwritten with the new OS, and marked as active. If something goes wrong (that can be detected/handled) it'll switch back to the other working partition. Since you don't know which is the active partition, you have to make the changes to both.

magnum, 2022-11-21:
ok thank you for the clarification.
1st of all, i'll try to have a stable pogo connection in order to test all the operations on files etc.
then i'll try to setup `xochitl.service`

it's better to symlink it in
`etc/systemd/system/xochitl.service.d/'
or 
`etc/systemd/system/multi-user.target.wants/`

thank you for your time and support, i'm learning a lot, considering my starting point... :-)

Eeems, 2022-11-21:
`multi-user.target.wants`
The other directory is just a place to override the contents of the file.

closed 2025-07-21 by Eeems