ddvk/remarkable2-recovery #18: Not see device USB HID v1.10 Device [Freescale SemiConductor Inc SE Blank ULT1]

Brucecarl, 2023-02-08:
Hi,
   I followed your steps and it seems did not work for me.I can not see msg:"device USB HID v1.10 Device [Freescale SemiConductor Inc SE Blank ULT1]".Can you help to find what's wrong?
[image]
[image]
[image]
[image]

Eeems, 2023-02-08:
How are you securing the pogo connector? Doesn't it have enough pressure?

Brucecarl, 2023-02-08:
> How are you securing the pogo connector? Doesn't it have enough pressure?
I use adhesive tape! I think the pogo connect is ok,because when I connect the pogo pin,the device can start itself.

alistair23, 2023-02-08:
Is your USB-C cable breaking carrying all of the pins?

Brucecarl, 2023-02-09:
> Is your USB-C cable carrying all of the pins?
Yeah

ddvk, 2023-02-10:
the pogo connection doesnt look stable

Brucecarl, 2023-02-10:
> the pogo connection doesnt look stable

I also tried to hold the pins with fingers heavielly(and use adhesive tape to secure connection),but it still not work...

wordpr3ss, 2023-04-21:
Any luck? I am in a similar situation. The device was bootlooping, showing "starting" then "restarting" and again.
I had everything setup according to the configuration, but didn't see anything in the dmesg.

BTW, I saw you are using a mac. so are you running a virtual linux inside? I thought only linux was supported.

mrares, 2023-04-24:
I have the same issue, I'm using spring-loaded pogo pins (I'm pretty confident in the connection and I tried jiggling the board, etc.), connected to a Raspberry Pi. The Rm2 starts booting in a loop when I connect the usb cable but nothing shows up in dmesg

ForkYeahh, 2023-07-27:
I'm having the same issue, did anyone manage to fix this issue? 

When I connect the remarkable v2 it does boot, but I in my `dmesg` log it does not show any USB device.

mrares, 2023-08-06:
Using the ***right kind of USB Cable!*** Solved it for me!

Make sure you're using a USB 3.1 cable before troubleshooting further.

ForkYeahh, 2023-08-10:
Using the right cable did not work for me, maybe my pogo connection is not stable enough. I also tried soldering on the pogo pins to make sure the connection is stable but still no luck. The rm2 boots when I connect it to my pc but still no USB devices are shown in the `dmesg -w` log. Does this exploit only work on certain versions? My rm2 is on version `3.4.1.1790`

mrares, 2023-08-10:
Are you sure you put the device in the debug mode by holding the button? The version shouldn't matter because this is a preboot thing where you're talking directly to the chip, the os doesn't boot at this time.

Sadly it *is* possible that something is fried. If you have an oscilloscope you could check the Data lines over the pogo pins to see if any signalling happens at all.

ForkYeahh, 2023-08-10:
> Are you sure you put the device in the debug mode by holding the button? The version shouldn't matter because this is a preboot thing where you're talking directly to the chip, the os doesn't boot at this time.

What do you mean hold the button? It boots automatically when I connect the cable so I thought I didn't have to hold the button, or am I wrong?

> Sadly it is possible that something is fried. If you have an oscilloscope you could check the Data lines over the pogo pins to see if any signalling happens at all.
I don't think anything is fried since the rm2 still works perfectly, but i'll try to check with an oscilloscope what happens over the pogo pins today

mrares, 2023-08-10:
> What do you mean hold the button? It boots automatically when I connect the cable so I thought I didn't have to hold the button, or am I wrong?

Sorry, I haven't done this in a while. You should check that you are in fact pulling down B8, you should check your USB breakout board and make sure you're actually connecting B8 to a ground via resistor (in my case I think this wasn't exactly as shown in the guide)

Then you should try with the USB-C cable in both orientations (the B8 line will swap when you plug in the cable upside-down)

> I don't think anything is fried since the rm2 still works perfectly, but i'll try to check with an oscilloscope what happens over the pogo pins today

The USB port that comes via the pogo pins can be fried without affecting the device, it had its own circuitry right behind the pogo pins.

ForkYeahh, 2023-08-10:
Thanks for the advice I hadn't thought of trying the USB-C in both orientations, I'll try that and check with the oscilloscope. Just out of my own curiosity tho, why do we need to pull down the B8 line? What purpose does it have, I don't see that mentioned in the readme besides that we need to do it

mrares, 2023-08-10:
> why do we need to pull down the B8 line?

Tbh I haven't pulled up the documentation to the freescale chip that the remarkable uses so please take this with a grain of salt!

The point of using this pulldown is to change the signal the processor is getting on some kind of special pin. Inside the device the pin is pulled up (to 3.3 or 5v) somehow so the signal registers "on" to the chip. 

What they did was make it possible for you to change this signal externally using the breakout board and resistor to "low". Without this facility you'd have to break open the device and physically solder into it, it would be a MESS.

When you do this pull-down and power the device the processor itself no longer boots the device with its regular OS but rather enters a completely different mode of operation where it behaves like a "development board". 

This mode allows you to upload firmware you want it to run, the custom firmware that the guide uses is actually a "hey look Ma I'm a USB stick now" firmware, so when you upload it it makes the device expose its internal storage to you as if it was an USB stick, from there on you can go around and ~break~fix things.

ForkYeahh, 2023-08-11:
Cool thanks for the explanation! 

I've monitored the pogo pins on the rm2 with an oscilloscope and the pins do still seem to be working, thereby also confirming that the soldering on the pogo pins works. Weirdly enough I'm still not seeing any logs in the `dmesg -w` output. I'm using ubuntu 22.04.01 LTS.

I've also tried both orientations of the USB-C cable making sure that the B8 line is correctly oriented. Not sure what I'm doing wrong at this point 🧐

Eeems, 2023-08-11:
Have you tried using a different port on your computer, or restarting it?

GijsGoudzwaard, 2023-08-11:
Yeah I even tried like 3 different computers 😅 different operating systems even so I'm guessing that's not the issue. Also tried like 10 different Micro usb cables and multiple USB-C cables

Eeems, 2023-08-11:
Do you see anything when connecting your reMarkable normally without pulling the pin down?

mrares, 2023-08-12:
Without anything connected, do you see any activity on the screen when you try to turn on the device?

After you turn off the device (keep power button pressed for a while) connecting the b8 line board should trigger a start without using the power button (and this is when you should see activity on the pogo pins)

ForkYeahh, 2023-08-14:
> Without anything connected, do you see any activity on the screen when you try to turn on the device?
You mean without connecting the rm2 to the pc and checking dmesg? Yeah the rm2 boots just fine and still works perfectly. Even when it is connected to the pc.

 > After you turn off the device (keep power button pressed for a while) connecting the b8 line board should trigger a start without using the power button (and this is when you should see activity on the pogo pins)
Tried this method and there is still no activity in the dmesg log...

For context this is my setup
[image]
[image]
[image]

I've also tried using pogo connectors to connect to the rm2 connectors.

Eeems, 2023-08-14:
I'm guessing in these pictures you are just using the device normally and not trying to get it into recovery mode? The reason I'm assuming this is because I see xochitl running on the screen. Unless this is just because it's stuck in that state due to not letting it get back to the normal boot process?

ForkYeahh, 2023-08-14:
The rm2 is not stuck in this state it works as it normally would. I am trying to get it into recovery mode but with no succes. Firstly I power off the device, then I connect the rm2 to my pc and pull down B8. Once connected the rm2 boots automatically until it is back on the xochitl screen. The screen you see on the picture is the screen you see once it has automatically booted, for testing purposes I've removed the pincode from the rm2.

mrares, 2023-08-14:
> Once connected the rm2 boots automatically

Here you're meaning connecting the b8 line, right?

When you do this (on start triggered by b8) the rm2 shouldn't be booting normally

ForkYeahh, 2023-08-14:
No once I connect the rm2 to my pc (through the micro usb breakout board) the rm2 boots normally

mrares, 2023-08-14:
Ah! I might be misremembering what the behavior should be but this makes sense.

The behavior you're describing points at the b8 line not being pulled down correctly

ForkYeahh, 2023-08-14:
Hm, I did use an 8k resistor since I didn't have a 10k resistor. I just now put a 8.2k and a 1.8k resistor in series to achieve 10k but still no luck 🧐

mrares, 2023-08-14:
Check the line with a scope (preferred) or multimeter, with and without the resistor, to ground (or what you think is ground). Consider there's something wrong witht the breakout board you're using, try another one, or another kind of board (I noticed yours isn't exactly the one in the how-to)

ForkYeahh, 2023-08-14:
I'll check the lines with an oscilloscope today or tomorrow to be sure. I already verified the ground points to be ground, but I'll check the B8 to make sure.

It could be that the breakout board does not work, though it looks like it should work to me

[image]
[image]

ForkYeahh, 2023-08-15:
I measured the B8 line and I see it being pulled down once I add my resistors. Also made sure that the GND i'm using is in fact GND. The board seems to be working...

ForkYeahh, 2023-08-29:
So... Apparently I measured everything and changed every cable like 10 times but the only thing I didn't think of was changing the USB-C cable 🤦 After changing it to the right USB-C cable that does have the B8 line it instantly worked. Thanks for the help nonetheless!

levensonblaine, 2024-06-24:
Had the same situation, the problem is in your USB-C cable that doesn't include data transfer through B8. You can fix it by using usb-c 3.2 gen 2 or male usb-c breakout board. You will see additional screen flashing (3 instead of 2). That means recovery mode that disables normal booting.
When you are not in recovery mode, your contact with pogo pins makes normal boot. If it has no reaction contacting pogo pins - you have problem with pogo pins. It has to make boot (without recovery mode) or mount to your system (recovery mode)