ddvk/remarkable2-recovery #21: OS info

N4S4, 2023-04-07:
Hi,

thanks for the beautiful instructions, even because customer support is the worse.
unfortunately i stopped the UI service by mistake so my remarkable is stuck on starting screen.
I know in the instructions says to use linux but by any chance do you know if linux wsl in windows works?

thank you

Eeems, 2023-04-07:
Did you stop, or disable the service? If you stopped it, just hold the power button for 10+ seconds, and then power it on again. If you disabled the service, you should still be able to SSH in if you have the password or setup a SSH key, at which point you can just enable and start the service again.

N4S4, 2023-04-07:
well i stopped and started another service which i installed via ssh but than i realized later that was not compatible....
 i do not have ssh pass anymore unfortunately

Eeems, 2023-04-07:
> I know in the instructions says to use linux but by any chance do you know if linux wsl in windows works?

In theory, you might be able to get it to work, but you'd have to jump through hoops to get the USB device passed through, and you won't have the immediate feedback with dmesg that you'd get by running linux natively. I'd recommend just putting ubuntu onto a usb drive and live booting that: https://ubuntu.com/tutorials/install-ubuntu-desktop#3-create-a-bootable-usb-stick

N4S4, 2023-04-07:
yeah i was thinking to boot a live distro and try...well I will try that definitely and revert back... thank you

Eeems, 2023-04-07:
What instructions were you following that lead to this?

N4S4, 2023-04-07:
it was during the installation of one of the launcher https://github.com/reHackable/awesome-reMarkable#take-special-care-if-you-are-using-a-remarkable-2  not sure which one was as i totally erased my memory for the nervous lol

Eeems, 2023-04-07:
Ah, so it looks like you missed this portion right above it: https://github.com/reHackable/awesome-reMarkable#write-down-your-ssh-password

N4S4, 2023-04-07:
indeed

Eeems, 2023-04-07:
I've opened https://github.com/reHackable/awesome-reMarkable/pull/172 to make the warning section stand out more.

N4S4, 2023-04-07:
Yep that is a great idea!

closed 2025-07-21 by Eeems