ddvk/remarkable2-recovery #22: My Thanks and my Experience

jatherrien, 2023-04-08:
Hi,

This isn't really an issue (I'll mark it as closed right away) but more a testimony for how these instructions saved my device. After updating my Remarkable to the latest version I tried reinstalling Toltec without checking that it supported the latest version. After butchering a repair over SSH the only way I could save it was through this recovery method (which succeeded). 

Some notes for other people - 
* I found that after connecting the resistor to trigger the recovery mode, I had to press the power button once and then wait for about 2 minutes before proceeding to try to get a USB connection. 
* Getting a stable USB connection is very, very hard without a Pogo connector. In my case I ended up having to solder directly on those pins, but 'next' time I'm not going to skip out on getting the right connector. 
* Make sure the device is actually fully powered off before starting the procedure - under Troubleshooting in the guide it suggests plugging into a charger to see the screen turn on, and then holding the power button for 15 seconds. If I tried skipping that I often had trouble getting into recovery.

Finally, for me to recover my particular issue I removed all the toltec and remux systemd services on both partitions. If you use a different screen switcher then make sure that's disabled too.

closed 2023-04-08 by jatherrien

Eeems, 2023-04-08:
I'd recommend opening a PR to add some of your notes to the readme, that way it'll be easier for people to find.

joncfoo, 2025-08-30:
I also would like to express my thanks!  This repo helped me recover my bricked RM2 after a faulty firmware upgrade.