ddvk/remarkable2-recovery #25: Which breakdown does this project aim to？

Lebesge, 2024-03-01:
Hey, guys. My rM 2 is always stuck at the  starting page and has no response/change when charging or connecting to my computer. So I want to know if this project could repair my device, many thanks.

Eeems, 2024-03-01:
If your device is still functional, and you just don't have UI and SSH access. Yes it can be used to restore it to a working state. If your device is failing to power on due to a hardware reason, no it's unlikely this will be able to do anything.

Lebesge, 2024-03-01:
> If your device is still functional, and you just don't have UI and SSH access. Yes it can be used to restore it to a working state. If your device is failing to power on due to a hardware reason, no it's unlikely this will be able to do anything.

What do you mean by "functional" and how to check that? I tend to believe the problem is not of hardware and may be caused by an incorrect manipulation when accessing SSH.

Lebesge, 2024-03-01:
> > > If your device is still functional, and you just don't have UI and SSH access. Yes it can be used to restore it to a working state. If your device is failing to power on due to a hardware reason, no it's unlikely this will be able to do anything.
> > 
> > What do you mean by "functional" and how to check that? I tend to believe the problem is not of hardware and may be caused by an incorrect manipulation when accessing SSH. It's frozen.
> 
> Does the screen flash at all when hard resetting the device? (Hold down power button for 10 seconds, and then press it for 1-2 seconds to power the device back on)

No flash.

Lebesge, 2024-03-01:
Maybe it doesn't work for my problem.

Eeems, 2024-03-02:
Have you tried charging it for 3-4 hours first on a slow charger or computer port?

Lebesge, 2024-03-02:
> Have you tried charging it for 3-4 hours first on a slow charger or computer port?

Yes. I tried it by nothing happened. Thanks for your answers.

Eeems, 2024-03-02:
> > Have you tried charging it for 3-4 hours first on a slow charger or computer port?
> 
> Yes. I tried it by nothing happened. Thanks for your answers.

Does your computer recognize that a device is connected?

Lebesge, 2024-03-02:
> > > Have you tried charging it for 3-4 hours first on a slow charger or computer port?
> > 
> > Yes. I tried it by nothing happened. Thanks for your answers.
> 
> Does your computer recognize that a device is connected?

No. When I connect it to my computer, my computer has no tone and I can't access it via SSH.

Eeems, 2024-03-02:
> > > > Have you tried charging it for 3-4 hours first on a slow charger or computer port?
> > > 
> > > Yes. I tried it by nothing happened. Thanks for your answers.
> > 
> > Does your computer recognize that a device is connected?
> 
> No. When I connect it to my computer, my computer has no tone and I can't access it via SSH.

It does sound like you will be unable to use this to recover your device.

Lebesge, 2024-03-02:
OK, thanks.

closed 2024-03-02 by Lebesge