ddvk/remarkable2-recovery #26: Device not starting with Pulldown resistor

MRUnbekanntOP, 2024-03-18:
I have the following Problem:
I originally Bricked my ReMarkable 2 by installing a chess mod, which brought my device into a boot loop when I tried to exit it. I also lost any direct ssh connection to the device, it just wasn’t connecting to anything. So, I ordered the Parts mentioned on this github page. Now my Problem: the device starts “normally” (entering the boot loop), when I connect nothing or just the USB pogo pins. But, when I connect the type-c Breakout board and pull down B8 (I used a 10k resistor from an Arduino set), The Device would not start, not even when I press the button. I have a male type-C breakout board, so the orientation of the port is not the Problem. Also, when I disconnect the Pulldown, the device enters the bootloop again after about 10s. This behavior is also independent from the Pogo pins (I also theoretically get a stable connection for them, but they do not affect this problem in any way). I also thought that this may just be a display setting in recovery mode, but the dmesg also does not show anything, neither with pulldown nor without.

Etn40ff, 2024-03-18:
> The Device would not start, not even when I press the button
what do you mean by that?

MRUnbekanntOP, 2024-03-18:
> > The Device would not start, not even when I press the button
> > what do you mean by that?

When I pull down b8, the boot loop stops and, no matter what I do, the display does not show any changes, no matter what i do. So, it shows the behavior as if it was turned off and can‘t be started

Etn40ff, 2024-03-18:
The display is not supposed to show any sign. You should see things happening in dmesg. If not check your pogo connection and the cable you are using

MRUnbekanntOP, 2024-03-19:
I just tried it again. When I connect the Pulldown resistor, the display changes of the Device stopped, but I did not get any  Reactions  on dmesg. I also tried different micro-USB-Cables. I am certain that my connection to the Pogo pins is solid, and I do not have an Oscilloscope.

Etn40ff, 2024-03-20:
Try things in this order: 
- take both the pogo pin and the resistor out of your remarkable
- hold the power button long enough to make sure that the device is off
- plug in the resistor
- connect the pogo pins (it is a good idea to connect the pogo pins before starting and only plug the USB cable in your PC at this stage)
- look at dmesg, your remarkable should appear now as FreeScale bla bla
- if not press the power button
- continue with the usual instructions

MRUnbekanntOP, 2024-03-20:
I did precisely what you described, but I do not get any outputs on dmesg.

MRUnbekanntOP, 2024-03-20:
@Etn40ff Thanks for your help. After your comment (specifically that there should be at least some output), I completely rebuilt my pogo pin connection and now I see the correct output. Again, thanks!

closed 2024-03-20 by MRUnbekanntOP

Etn40ff, 2024-03-20:
I think that comment was due to @Eeems. Either way I am glad you fixed it

MRUnbekanntOP, 2024-03-21:
Oh right, I missed that when writing the other comment. @Eeems Thank you