ddvk/remarkable2-recovery #31: Anything else i can do?

Lunari-sol, 2024-06-07:
Hi there
I have followed the procedure here, and I can get into the file system of my remarkable consistently. I have done the symbolic link creation recommended in the other issues, I've done the dd rewrite of the active partition using the previous partition, I've even tried using an official release (3.3.2.1666) to dd into the active partition, I have my ssh key, I backed up all my files, my remarkable does flash whenever i start it so i think the hardware is fine, but it still just sits on the "remarkable is starting page". 

I cannot get a recent log file as well, the most recent being from the partition that was working fine right before I bricked it and no new log files have been generated since after I bricked it (two days ago at 11am), on either partition.

I had downgraded so i could install toltec and koreader and remux, and it worked and i used it all fine, but I found them not fulfilling what i wanted, so I uninstalled them using toltecs guide, and then tried to upgrade the remarkable back to 3.9 using remarkable-upgrade, switch.sh wasnt working so i tried to just type the commands from the file into the kernal (I didnt and still dont fully know what I'm doing) and after that I restarted it and it bricked. 

Im thinking maybe I changed the environmental variable incorrectly and that's why its not working? I dont even know how I would check that to be the case, or if thats even a thing that can happen. The tablet isnt recognized by my home computer when i plug it in usb, but like i said i can get into the file system fine using this guide. 

I also did have a dmesg output of "ext4_validate_block_bitmap:421: comm ext4lazyinit: bg 32: bad block bitmap checksum" at some point halfway through my troubleshooting today, so i ran fsck and haven't seen the same error since. 

Id appreciate any advice, even if its "this is not fixable", I just am at a brick wall and I am not smart enough to even guess further at what might be the issue.

Lunari-sol, 2024-06-07:
Yes - I cannot ssh in via usb at all, my computer doesn't recognize the remarkable as even being plugged in

I have turned both partitions into stock 3.9 and it still is stuck on the 'is starting' screen, no visible changes in its behavior

Eeems, 2024-06-07:
Can you verify that the partitions match the image file, as well as run a fsck on both the partitions and the image file.

I'm assuming you went with 3.9.5.2026. When I extract the image from the upgrade file on my machine, the sha256sum is `a3e8c0e068290828daf6a75bdbdb91ccdc9efeeb31f83c81e5d6d1b69b64e052` and fsck comes back clean:
```shell
eeems@reaper  ~  fsck /tmp/3.9.5.2026.img 
fsck from util-linux 2.40.1
e2fsck 1.47.1 (20-May-2024)
/tmp/3.9.5.2026.img: clean, 8048/34816 files, 218520/276480 blocks
```

Mounting the image as a loopback provides the following file listing (`ls -l`) at the root folder:
```
total 51
drwxr-xr-x  2 root root  3072 Jan 16 06:39 bin
drwxr-xr-x  2 root root  1024 Jan 16 06:39 boot
drwxr-xr-x  2 root root  1024 Jan 16 06:39 dev
drwxr-xr-x 26 root root  3072 Feb 12 05:29 etc
drwxr-xr-x  3 root root  1024 Jan 16 06:39 home
drwxr-xr-x  8 root root  3072 Feb 12 05:29 lib
drwx------  2 root root 12288 Jan 16 08:06 lost+found
drwxr-xr-x  2 root root  1024 Jan 16 06:39 media
drwxr-xr-x  2 root root  1024 Jan 16 06:39 mnt
-rwxr-xr-x  1 root root  6889 Jan 16 06:39 postinst
-rwxr-xr-x  1 root root  1323 Jan 16 06:39 postinst-waveform
dr-xr-xr-x  2 root root  1024 Jan 16 06:39 proc
drwxr-xr-x  2 root root  1024 Jan 16 06:39 run
drwxr-xr-x  2 root root  3072 Jan 16 06:39 sbin
drwxr-xr-x  2 root root  1024 Jan 16 06:39 srv
dr-xr-xr-x  2 root root  1024 Jan 16 06:39 sys
drwxrwxrwt  2 root root  1024 Jan 16 06:39 tmp
-rwxr-xr-x  1 root root  4296 Jan 16 06:39 uboot-postinst
-rw-r--r--  1 root root    31 Jan 16 06:39 uboot-version
drwxr-xr-x 10 root root  1024 Feb 12 05:29 usr
drwxr-xr-x  9 root root  1024 Jan 16 06:39 var
```
As well as the following listing for the boot folder:
```
total 6676
-rw-r--r-- 1 root root  647212 Jan 16 06:39 u-boot.imx
lrwxrwxrwx 1 root root      62 Jan 16 06:39 u-boot.imx-zero-sugar_config -> u-boot-zero-sugar_config-v2018.03+gitAUTOINC+47c91918aa-r0.imx
-rw-r--r-- 1 root root  647212 Jan 16 06:39 u-boot-zero-sugar_config-v2018.03+gitAUTOINC+47c91918aa-r0.imx
-rw-r--r-- 1 root root   47554 Jan 16 06:39 zero-sugar.dtb
-rw-r--r-- 1 root root   32111 Jan 16 06:39 zero-sugar-kdump.dtb
lrwxrwxrwx 1 root root      26 Jan 16 06:39 zImage -> zImage-5.4.70-v1.3.4-rm11x
-rw-r--r-- 1 root root 5583944 Jan 16 06:39 zImage-5.4.70-v1.3.4-rm11x
```
Some sha256sums of important files for the boot process:
```
8170f604ba2eee0da8141b90df91ff2e8a47b938802d383e6dcd66ee2e4f6682  u-boot-zero-sugar_config-v2018.03+gitAUTOINC+47c91918aa-r0.imx
05d1b8c283c51dccea2e8616fcd63dd6b96b1ef2bf9f62bd1c9c6c43c94d65ae  zImage-5.4.70-v1.3.4-rm11x
7cc181c5b3eeafe98823a70fe0cc30265c31798ae1e9361263ba501e7785cdc1  zero-sugar.dtb

```

Lunari-sol, 2024-06-07:
Okay for the checksums i have
`a3e8c0e068290828daf6a75bdbdb91ccdc9efeeb31f83c81e5d6d1b69b64e052 /3.9.5.2026.img`

`8170f604ba2eee0da8141b90df91ff2e8a47b938802d383e6dcd66ee2e4f6682 /boot/u-boot-zero-sugar_config-v2018.03+gitAUTOINC+47c91918aa-r0.imx` 

`05d1b8c283c51dccea2e8616fcd63dd6b96b1ef2bf9f62bd1c9c6c43c94d65ae /boot/zImage-5.4.70-v1.3.4-rm11x`

`7cc181c5b3eeafe98823a70fe0cc30265c31798ae1e9361263ba501e7785cdc1  /boot/zero-sugar.dtb`

Which are the same as the ones you sent

Mounting the image as a loop, I got the same 

`total 51
drwxr-xr-x  2 root root  3072 Jan 16 05:39 bin
drwxr-xr-x  2 root root  1024 Jan 16 05:39 boot
drwxr-xr-x  2 root root  1024 Jan 16 05:39 dev
drwxr-xr-x 26 root root  3072 Feb 12 04:29 etc
drwxr-xr-x  3 root root  1024 Jan 16 05:39 home
drwxr-xr-x  8 root root  3072 Feb 12 04:29 lib
drwx------  2 root root 12288 Jan 16 07:06 lost+found
drwxr-xr-x  2 root root  1024 Jan 16 05:39 media
drwxr-xr-x  2 root root  1024 Jan 16 05:39 mnt
-rwxr-xr-x  1 root root  6889 Jan 16 05:39 postinst
-rwxr-xr-x  1 root root  1323 Jan 16 05:39 postinst-waveform
dr-xr-xr-x  2 root root  1024 Jan 16 05:39 proc
drwxr-xr-x  2 root root  1024 Jan 16 05:39 run
drwxr-xr-x  2 root root  3072 Jan 16 05:39 sbin
drwxr-xr-x  2 root root  1024 Jan 16 05:39 srv
dr-xr-xr-x  2 root root  1024 Jan 16 05:39 sys
drwxrwxrwt  2 root root  1024 Jan 16 05:39 tmp
-rwxr-xr-x  1 root root  4296 Jan 16 05:39 uboot-postinst
-rw-r--r--  1 root root    31 Jan 16 05:39 uboot-version
drwxr-xr-x 10 root root  1024 Feb 12 04:29 usr
drwxr-xr-x  9 root root  1024 Jan 16 05:39 var`

The only differences being the times

And for fsck i got:
`sol@raspberrypi:~ $ fsck /codexctl/3.9.5.2026.img 
fsck from util-linux 2.38.1
e2fsck 1.47.0 (5-feb-2023)
/codexctl/3.9.5.2026.img: clean, 8048/34816 files, 218520/276480 blocks`

`sol@raspberrypi:~ $ fsck /dev/sda2 
fsck from util-linux 2.38.1
e2fsck 1.47.0 (5-feb-2023)
/dev/sda2: clean, 8048/34816 files, 218520/276480 blocks`

`sol@raspberrypi:~ $ fsck /codexctl/3.9.5.2026.img 
fsck from util-linux 2.38.1
e2fsck 1.47.0 (5-feb-2023)
/dev/sda3: clean, 8048/34816 files, 218520/276480 blocks`

Lunari-sol, 2024-06-07:
I realize i should have checked the sha256sum for the boot files in each partition, and in both they are the same as above and as yours

Eeems, 2024-06-07:
The times being different would just be your timezone vs mine from what I can tell. 

Have you compared the partitions to the image yet to confirm they match?

Another thing, you said you can't get the most recent log file? To confirm, you were inspecting the systemd-journal archive on the root partition right? If not, that's what you would need to look at for the most recent log file. You can open it with journalctl on the host machine, there is a flag to point it at a different directory than the system journal.

If the archive doesn't contain anything new, we can probably assume that something is keeping it from booting the kernel properly, which is very odd. It could be something was corrupted for the uboot on the device, although I'm less likely to think that's the case, as uboot needs to be working to get recovery working. If you can't get a log file, we might need to investigate the boot partition contents, which I can help you rebuild from my device if necessary.

Lunari-sol, 2024-06-07:
Ive been using journalctl in the /var/log/journal/ of each partition, thats where i saw others get pointed to look, before they contained log files that ended ~2 minutes before i bricked the device, and now both folders are empty, which is what led me to conclude im not getting the most recent. If the systemd-journal is something different (i apologize i dont know the terminology well) then i can look for it

i just went through again and i couldnt find any other system logs

Eeems, 2024-06-07:
That would be the journal, if there is nothing there, I would assume systemd is not starting up, or failing before journald gets to write anything back to disk.

I'm poking at uboot on the device, as I haven't looked too closely at how it's configured before, and we don't expose what I would think needs to update over recovery. I'm asking ddvk some questions to see if he has any suggestions.

> Im thinking maybe I changed the environmental variable incorrectly and that's why its not working? I dont even know how I would check that to be the case, or if thats even a thing that can happen. [...]

If you modified `uboot.env` on the first partition, this may be an issue. What does your file currently contain? I would recommend grabbing it with `xxd` to see both the hex and the text characters, as there are non-printable bytes.

Mine is 8192 bytes long, with the following content:
```
00000000: d3e6 3ab1 6163 7469 7665 5f70 6172 7469  ..:.active_parti
00000010: 7469 6f6e 3d32 0061 6c74 626f 6f74 636d  tion=2.altbootcm
00000020: 643d 6563 686f 2052 756e 6e69 6e67 2066  d=echo Running f
00000030: 726f 6d20 6661 6c6c 6261 636b 2072 6f6f  rom fallback roo
00000040: 742e 2e2e 3b20 7365 7465 6e76 206d 6d63  t...; setenv mmc
00000050: 7061 7274 2024 7b66 616c 6c62 6163 6b5f  part ${fallback_
00000060: 7061 7274 6974 696f 6e7d 3b20 7365 7465  partition}; sete
00000070: 6e76 2062 6f6f 7461 7267 7320 636f 6e73  nv bootargs cons
00000080: 6f6c 653d 247b 636f 6e73 6f6c 657d 2c24  ole=${console},$
00000090: 7b62 6175 6472 6174 657d 2072 6f6f 743d  {baudrate} root=
000000a0: 2f64 6576 2f6d 6d63 626c 6b32 7024 7b66  /dev/mmcblk2p${f
000000b0: 616c 6c62 6163 6b5f 7061 7274 6974 696f  allback_partitio
000000c0: 6e7d 2072 6f6f 7477 6169 7420 726f 6f74  n} rootwait root
000000d0: 6673 7479 7065 3d65 7874 3420 7277 2071  fstype=ext4 rw q
000000e0: 7569 6574 2070 616e 6963 3d32 3020 7379  uiet panic=20 sy
000000f0: 7374 656d 642e 6372 6173 685f 7265 626f  stemd.crash_rebo
00000100: 6f74 2024 7b6c 6f67 6172 6773 7d3b 2072  ot ${logargs}; r
00000110: 756e 206d 6d63 626f 6f74 3b00 6261 7564  un mmcboot;.baud
00000120: 7261 7465 3d31 3135 3230 3000 626f 6f74  rate=115200.boot
00000130: 636d 643d 6966 2074 6573 7420 2120 2d65  cmd=if test ! -e
00000140: 206d 6d63 2030 3a31 2075 626f 6f74 2e65   mmc 0:1 uboot.e
00000150: 6e76 3b20 7468 656e 2073 6176 6565 6e76  nv; then saveenv
00000160: 3b20 6669 3b20 7275 6e20 6d65 6d62 6f6f  ; fi; run memboo
00000170: 743b 2072 756e 206d 6d63 6172 6773 3b20  t; run mmcargs; 
00000180: 7365 7465 6e76 206d 6d63 7061 7274 2024  setenv mmcpart $
00000190: 7b61 6374 6976 655f 7061 7274 6974 696f  {active_partitio
000001a0: 6e7d 3b20 7275 6e20 6d6d 6362 6f6f 743b  n}; run mmcboot;
000001b0: 2065 6368 6f20 5741 524e 3a20 756e 6162   echo WARN: unab
000001c0: 6c65 2074 6f20 626f 6f74 2066 726f 6d20  le to boot from 
000001d0: 6569 7468 6572 2052 414d 206f 7220 654d  either RAM or eM
000001e0: 4d43 3b20 7365 7465 6e76 2075 7067 7261  MC; setenv upgra
000001f0: 6465 5f61 7661 696c 6162 6c65 2031 3b20  de_available 1; 
00000200: 7361 7665 656e 763b 2072 6573 6574 3b20  saveenv; reset; 
00000210: 0062 6f6f 7464 656c 6179 3d30 0062 6f6f  .bootdelay=0.boo
00000220: 746c 696d 6974 3d31 0063 6f6e 736f 6c65  tlimit=1.console
00000230: 3d74 7479 6d78 6330 0065 7468 7072 696d  =ttymxc0.ethprim
00000240: 653d 6574 6830 0066 616c 6c62 6163 6b5f  e=eth0.fallback_
00000250: 7061 7274 6974 696f 6e3d 3300 6661 7374  partition=3.fast
00000260: 626f 6f74 5f64 6576 3d6d 6d63 3000 6664  boot_dev=mmc0.fd
00000270: 745f 6164 6472 3d30 7838 3330 3030 3030  t_addr=0x8300000
00000280: 3000 6664 745f 6669 6c65 3d2f 626f 6f74  0.fdt_file=/boot
00000290: 2f7a 6572 6f2d 7375 6761 722e 6474 6200  /zero-sugar.dtb.
000002a0: 6664 7463 6f6e 7472 6f6c 6164 6472 3d62  fdtcontroladdr=b
000002b0: 6466 3261 6337 3000 696d 6167 653d 2f62  df2ac70.image=/b
000002c0: 6f6f 742f 7a49 6d61 6765 0069 705f 6479  oot/zImage.ip_dy
000002d0: 6e3d 7965 7300 6c6f 6164 6164 6472 3d30  n=yes.loadaddr=0
000002e0: 7838 3038 3030 3030 3000 6c6f 6164 6664  x80800000.loadfd
000002f0: 743d 6578 7434 6c6f 6164 206d 6d63 2024  t=ext4load mmc $
00000300: 7b6d 6d63 6465 767d 3a24 7b6d 6d63 7061  {mmcdev}:${mmcpa
00000310: 7274 7d20 247b 6664 745f 6164 6472 7d20  rt} ${fdt_addr} 
00000320: 247b 6664 745f 6669 6c65 7d00 6c6f 6164  ${fdt_file}.load
00000330: 696d 6167 653d 6578 7434 6c6f 6164 206d  image=ext4load m
00000340: 6d63 2024 7b6d 6d63 6465 767d 3a24 7b6d  mc ${mmcdev}:${m
00000350: 6d63 7061 7274 7d20 247b 6c6f 6164 6164  mcpart} ${loadad
00000360: 6472 7d20 247b 696d 6167 657d 006c 6f67  dr} ${image}.log
00000370: 6172 6773 3d73 7973 7465 6d64 2e73 686f  args=systemd.sho
00000380: 775f 7374 6174 7573 3d30 206c 6f67 6c65  w_status=0 logle
00000390: 7665 6c3d 3020 6372 6173 686b 6572 6e65  vel=0 crashkerne
000003a0: 6c3d 3332 4d00 6d65 6d62 6f6f 743d 6563  l=32M.memboot=ec
000003b0: 686f 2042 6f6f 7469 6e67 2066 726f 6d20  ho Booting from 
[... 429 lines cut ...]
00001e90: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001ea0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001eb0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001ec0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001ed0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001ee0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001ef0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
[... cut ...]
00001f90: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001fa0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001fb0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001fc0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001fd0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001fe0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001ff0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
```
[uboot.zip](https://github.com/user-attachments/files/15745719/uboot.zip)

Lunari-sol, 2024-06-07:
I would like to say thank you for containing to help me, especially as it seems I f'd up pretty bad, I really appreciate it

Here is the xxd output for my uboot.env, it is noticibly different, and as far as i can tell, a bit longer 
I do not know how to see how many bytes 
```
00000000: f642 e62c 616c 7462 6f6f 7463 6d64 3d65  .B.,altbootcmd=e
00000010: 6368 6f20 5275 6e6e 696e 6720 6672 6f6d  cho Running from
00000020: 2066 616c 6c62 6163 6b20 726f 6f74 2e2e   fallback root..
00000030: 2e3b 2073 6574 656e 7620 6d6d 6370 6172  .; setenv mmcpar
00000040: 7420 247b 6661 6c6c 6261 636b 5f70 6172  t ${fallback_par
00000050: 7469 7469 6f6e 7d3b 2073 6574 656e 7620  tition}; setenv 
00000060: 626f 6f74 6172 6773 2063 6f6e 736f 6c65  bootargs console
00000070: 3d24 7b63 6f6e 736f 6c65 7d2c 247b 6261  =${console},${ba
00000080: 7564 7261 7465 7d20 726f 6f74 3d2f 6465  udrate} root=/de
00000090: 762f 6d6d 6362 6c6b 3270 247b 6661 6c6c  v/mmcblk2p${fall
000000a0: 6261 636b 5f70 6172 7469 7469 6f6e 7d20  back_partition} 
000000b0: 726f 6f74 7761 6974 2072 6f6f 7466 7374  rootwait rootfst
000000c0: 7970 653d 6578 7434 2072 7720 7175 6965  ype=ext4 rw quie
000000d0: 7420 7061 6e69 633d 3230 2073 7973 7465  t panic=20 syste
000000e0: 6d64 2e63 7261 7368 5f72 6562 6f6f 7420  md.crash_reboot 
000000f0: 247b 6c6f 6761 7267 737d 3b20 7275 6e20  ${logargs}; run 
00000100: 6d6d 6362 6f6f 743b 0062 6175 6472 6174  mmcboot;.baudrat
00000110: 653d 3131 3532 3030 0062 6f6f 7461 7267  e=115200.bootarg
00000120: 733d 636f 6e73 6f6c 653d 7474 796d 7863  s=console=ttymxc
00000130: 302c 3131 3532 3030 2072 6f6f 743d 2f64  0,115200 root=/d
00000140: 6576 2f6d 6d63 626c 6b32 7020 726f 6f74  ev/mmcblk2p root
00000150: 7761 6974 2072 6f6f 7466 7374 7970 653d  wait rootfstype=
00000160: 6578 7434 2072 7720 7175 6965 7420 7061  ext4 rw quiet pa
00000170: 6e69 633d 3230 2073 7973 7465 6d64 2e63  nic=20 systemd.c
00000180: 7261 7368 5f72 6562 6f6f 7420 7379 7374  rash_reboot syst
00000190: 656d 642e 7368 6f77 5f73 7461 7475 733d  emd.show_status=
000001a0: 3020 6c6f 676c 6576 656c 3d30 2063 7261  0 loglevel=0 cra
000001b0: 7368 6b65 726e 656c 3d33 324d 0062 6f6f  shkernel=32M.boo
000001c0: 7463 6d64 3d69 6620 7465 7374 2021 202d  tcmd=if test ! -
000001d0: 6520 6d6d 6320 303a 3120 7562 6f6f 742e  e mmc 0:1 uboot.
000001e0: 656e 763b 2074 6865 6e20 7361 7665 656e  env; then saveen
000001f0: 763b 2066 693b 2072 756e 206d 656d 626f  v; fi; run membo
00000200: 6f74 3b20 7275 6e20 6d6d 6361 7267 733b  ot; run mmcargs;
00000210: 2073 6574 656e 7620 6d6d 6370 6172 7420   setenv mmcpart 
00000220: 247b 6163 7469 7665 5f70 6172 7469 7469  ${active_partiti
00000230: 6f6e 7d3b 2072 756e 206d 6d63 626f 6f74  on}; run mmcboot
00000240: 3b20 6563 686f 2057 4152 4e3a 2075 6e61  ; echo WARN: una
00000250: 626c 6520 746f 2062 6f6f 7420 6672 6f6d  ble to boot from
00000260: 2065 6974 6865 7220 5241 4d20 6f72 2065   either RAM or e
00000270: 4d4d 433b 2073 6574 656e 7620 7570 6772  MMC; setenv upgr
00000280: 6164 655f 6176 6169 6c61 626c 6520 313b  ade_available 1;
00000290: 2073 6176 6565 6e76 3b20 7265 7365 743b   saveenv; reset;
000002a0: 2000 626f 6f74 636f 756e 743d 3437 0062   .bootcount=47.b
000002b0: 6f6f 7464 656c 6179 3d30 0062 6f6f 746c  ootdelay=0.bootl
000002c0: 696d 6974 3d31 0063 6f6e 736f 6c65 3d74  imit=1.console=t
000002d0: 7479 6d78 6330 0065 7468 7072 696d 653d  tymxc0.ethprime=
000002e0: 6574 6830 0066 6173 7462 6f6f 745f 6465  eth0.fastboot_de
000002f0: 763d 6d6d 6330 0066 6474 5f61 6464 723d  v=mmc0.fdt_addr=
00000300: 3078 3833 3030 3030 3030 0066 6474 5f66  0x83000000.fdt_f
00000310: 696c 653d 2f62 6f6f 742f 7a65 726f 2d73  ile=/boot/zero-s
00000320: 7567 6172 2e64 7462 0066 6474 636f 6e74  ugar.dtb.fdtcont
00000330: 726f 6c61 6464 723d 6264 6632 6163 3730  roladdr=bdf2ac70
00000340: 0069 6d61 6765 3d2f 626f 6f74 2f7a 496d  .image=/boot/zIm
00000350: 6167 6500 6970 5f64 796e 3d79 6573 006c  age.ip_dyn=yes.l
00000360: 6f61 6461 6464 723d 3078 3830 3830 3030  oadaddr=0x808000
00000370: 3030 006c 6f61 6466 6474 3d65 7874 346c  00.loadfdt=ext4l
00000380: 6f61 6420 6d6d 6320 247b 6d6d 6364 6576  oad mmc ${mmcdev
00000390: 7d3a 247b 6d6d 6370 6172 747d 2024 7b66  }:${mmcpart} ${f
000003a0: 6474 5f61 6464 727d 2024 7b66 6474 5f66  dt_addr} ${fdt_f
000003b0: 696c 657d 006c 6f61 6469 6d61 6765 3d65  ile}.loadimage=e
000003c0: 7874 346c 6f61 6420 6d6d 6320 247b 6d6d  xt4load mmc ${mm
000003d0: 6364 6576 7d3a 247b 6d6d 6370 6172 747d  cdev}:${mmcpart}
000003e0: 2024 7b6c 6f61 6461 6464 727d 2024 7b69   ${loadaddr} ${i
000003f0: 6d61 6765 7d00 6c6f 6761 7267 733d 7379  mage}.logargs=sy
00000400: 7374 656d 642e 7368 6f77 5f73 7461 7475  stemd.show_statu
[... 425 lines cut ...]
00001ea0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001eb0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001ec0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001ed0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001ee0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001ef0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001f00: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001f10: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001f20: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001f30: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001f40: 0000 0000 0000 0000 0000 0000 0000 0000  ................
[... cut ...]
00001fa0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001fb0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001fc0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001fd0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001fe0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00001ff0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
```
This is both .env files from my boot folder, just in case
[bootenv.zip](https://github.com/user-attachments/files/15745761/bootenv.zip)

Eeems, 2024-06-07:
`ls -l` has the size in bytes in the column before the date.

Looking at the contents of your `uboot.env`, it does appear to be corrupted. It appears to be missing the `active_partion` variable. You could try replacing your `uboot.env` file with mine to see if that helps.

Lunari-sol, 2024-06-07:
That fixed it!

Lunari-sol, 2024-06-07:
i was scared to try it just in case, but yeah just replacing my uboot with yours has it working normally again, in version 3.9

Lunari-sol, 2024-06-07:
sorry for the three comments, you are amazing! thank you so so so so much!

closed 2024-06-07 by Lunari-sol

Eeems, 2024-06-07:
> i was scared to try it just in case, but yeah just replacing my uboot with yours has it working normally again, in version 3.9

Technically, you replaced your `uboot.env`, which is just variables uboot uses to inform it how to boot. Your uboot wasn't actually modified. Although that's what I was exploring looking into next.

> sorry for the three comments, you are amazing! thank you so so so so much!

No worries :) You can edit comments though in the future to avoid comment spam.

I'm glad that this got things working for you! I'll have to explore adding some more documentation based on this.