ddvk/remarkable2-recovery #34: Help to recover my son Remarkable 2 tablet

M3TALZ3R0, 2024-09-11:
Hello, can someone help me please? My 11 year old son's remarkable tablet 2 is locked and he can't remember the password. It was a gift from my brother to my son, but he lost his temporary email account and Remarkable can't help us if we don't have access to email. He's in art school and we live in a third world country and I can't afford to buy another tablet. I've tried everything but the tablet won't start by itself, it always turns on and I'm stuck on "you should see a new USB HID v1.10 device [Freescale SemiConductor Inc SE Blank ULT1]".

I can't get it to work. If anyone can help me with a video or more detailed instructions on terminal commands from the beginning please, I'm just a Linux newbie.

Eeems, 2024-09-11:
What do you see in the dmesg output when you connect it?
Does the device have a charge?
Is the USB cable the one that came with the device?

M3TALZ3R0, 2024-09-11:
What do you see in the dmesg output when you connect it?

> Here's what I see when I power on the remarkable with the pogo and the usb breakout connector:

> remarkable@remarkable-VirtualBox:~$ dmesg -w
[    0.000000] Linux version 6.8.0-44-generic (buildd@lcy02-amd64-082) (x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #44-Ubuntu SMP PREEMPT_DYNAMIC Tue Aug 13 13:35:26 UTC 2024 (Ubuntu 6.8.0-44.44-generic 6.8.12)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.0-44-generic root=UUID=37f12093-714d-4f3b-a149-70f1a2cb2a71 ro quiet splash
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000dffeffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dfff0000-0x00000000dfffffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000011fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000001] kvm-clock: using sched offset of 935957722566 cycles
[    0.000003] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000006] tsc: Detected 2304.006 MHz processor
[    0.001880] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001884] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001888] last_pfn = 0x120000 max_arch_pfn = 0x400000000
[    0.001901] MTRRs disabled by BIOS
[    0.001903] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001911] last_pfn = 0xdfff0 max_arch_pfn = 0x400000000
[    0.001955] found SMP MP-table at [mem 0x0009fff0-0x0009ffff]
[    0.002096] RAMDISK: [mem 0x302a9000-0x3414bfff]
[    0.002423] ACPI: Early table checksum verification disabled
[    0.002426] ACPI: RSDP 0x00000000000E0000 000024 (v02 VBOX  )
[    0.002430] ACPI: XSDT 0x00000000DFFF0030 00003C (v01 VBOX   VBOXXSDT 00000001 ASL  00000061)
[    0.002435] ACPI: FACP 0x00000000DFFF00F0 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[    0.002440] ACPI: DSDT 0x00000000DFFF0640 002353 (v02 VBOX   VBOXBIOS 00000002 INTL 20100528)
[    0.002443] ACPI: FACS 0x00000000DFFF0200 000040
[    0.002475] ACPI: FACS 0x00000000DFFF0200 000040
[    0.002480] ACPI: APIC 0x00000000DFFF0240 00008C (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
[    0.002483] ACPI: SSDT 0x00000000DFFF02D0 00036C (v01 VBOX   VBOXCPUT 00000002 INTL 20100528)
[    0.002486] ACPI: Reserving FACP table memory at [mem 0xdfff00f0-0xdfff01e3]
[    0.002487] ACPI: Reserving DSDT table memory at [mem 0xdfff0640-0xdfff2992]
[    0.002510] ACPI: Reserving FACS table memory at [mem 0xdfff0200-0xdfff023f]
[    0.002510] ACPI: Reserving FACS table memory at [mem 0xdfff0200-0xdfff023f]
[    0.002511] ACPI: Reserving APIC table memory at [mem 0xdfff0240-0xdfff02cb]
[    0.002511] ACPI: Reserving SSDT table memory at [mem 0xdfff02d0-0xdfff063b]
[    0.003745] No NUMA configuration found
[    0.003747] Faking a node at [mem 0x0000000000000000-0x000000011fffffff]
[    0.003768] NODE_DATA(0) allocated [mem 0x11ffd1000-0x11fffbfff]
[    0.004225] Zone ranges:
[    0.004226]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.004229]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[    0.004230]   Normal   [mem 0x0000000100000000-0x000000011fffffff]
[    0.004231]   Device   empty
[    0.004232] Movable zone start for each node
[    0.004233] Early memory node ranges
[    0.004233]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.004235]   node   0: [mem 0x0000000000100000-0x00000000dffeffff]
[    0.004236]   node   0: [mem 0x0000000100000000-0x000000011fffffff]
[    0.004237] Initmem setup node 0 [mem 0x0000000000001000-0x000000011fffffff]
[    0.004241] On node 0, zone DMA: 1 pages in unavailable ranges
[... 696 lines cut ...]
[ 7340.628056] usb 1-1: Product: USB Tablet
[ 7340.628057] usb 1-1: Manufacturer: VirtualBox
[ 7340.854753] input: VirtualBox USB Tablet as /devices/pci0000:00/0000:00:06.0/usb1/1-1/1-1:1.0/0003:80EE:0021.0002/input/input8
[ 7340.857794] hid-generic 0003:80EE:0021.0002: input,hidraw0: USB HID v1.10 Mouse [VirtualBox USB Tablet] on usb-0000:00:06.0-1/input0
[ 7344.710550] e1000: enp0s3 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[ 7870.679480] usb 1-1: USB disconnect, device number 3
[ 7870.794650] workqueue: hub_event hogged CPU for >10000us 4 times, consider switching to WQ_UNBOUND
[ 7871.320457] usb 1-1: new full-speed USB device number 4 using ohci-pci
[... cut ...]
[ 7871.700379] input: VirtualBox USB Tablet as /devices/pci0000:00/0000:00:06.0/usb1/1-1/1-1:1.0/0003:80EE:0021.0003/input/input9
[ 7871.702455] hid-generic 0003:80EE:0021.0003: input,hidraw0: USB HID v1.10 Mouse [VirtualBox USB Tablet] on usb-0000:00:06.0-1/input0
[ 7872.135190] e1000: enp0s3 NIC Link is Down
[ 7872.136595] e1000 0000:00:03.0 enp0s3: Reset adapter
[ 7874.429669] e1000: enp0s3 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX

Does the device have a charge?

> Yes, 100% full charge.

Is the USB cable the one that came with the device?

> Yes, orginal cable and works perfectly with other devices.

M3TALZ3R0, 2024-09-11:
I'm using a virtual box machine to emulate Ubuntu with all my drivers installed and working perfectly with other usb devices (like pendrives and phones).

I don't know if this has anything to do with the issue.

M3TALZ3R0, 2024-09-11:
[image]
[image]
[image]
[image]

Eeems, 2024-09-11:
> I'm using a virtual box machine to emulate Ubuntu with all my drivers installed and working perfectly with other usb devices (like pendrives and phones).
> 
> I don't know if this has anything to do with the issue.

As I'm not sure how you have usb passthrough configured, and I'm not sure how virtualbox handles it, this could be an issue.

The dmesg output you posted appears to be more than just what is added when you connect/start the device, which is too much for me to really go through. Can you get me just the output from when you connect the device and start it into recovery?

Could you also get a picture showing how you are connecting to the pogo pins, the image you have doesn't really show how the connection is made. My guess is that it's not a good connection from what I can infer from the image.

M3TALZ3R0, 2024-09-18:
Sorry for the late reply. I was trying to find a 5 pin pogo connector to get a better connection. I've tried but still the same result.
Also I got **Ubuntu 24.04.1 LTS** installed in my laptop just in case.
Here's the pic you asked for:

[image]
[image]
[image]

The VBUS is on the **YELLOW CABLE** near the USB-C connector and the **RED CABLE** is on the GND connector.

Also I did everything:

I've installed [libusb (1.0)] in order to install the [imx_usb_loader]

Tried to install [imx_usb_loader]:
Then I got this error while I was trying to install it:
"E: Unable to locate package git"

I use this command to update the package off-line cache:
$ sudo apt-get update
$ sudo apt-get upgrade

Installed [imx_usb_loader]:

Run the command "dmesg -w"
Got this error: "dmesg: read kernel buffer failed: Permission denied"

Fix it with the command:
% sudo sysctl kernel.dmesg_restrict=0
kernel.dmesg_restrict = 0

Run the command "dmesg -w" again.

After a lot of text, the last 4 lines was:

[41002.944018] xhci_hcd 0000:01:00.2: xHC error in resume, USBSTS 0x401, Reinit
[41002.944023] usb usb3: root hub lost power or was reset
[41002.944024] usb usb4: root hub lost power or was reset
[41248.503737] audit: type=1400 audit(1726661999.685:114): apparmor="DENIED" operation="change_onexec" class="file" info="label not found" error=-2 profile="unconfined" name="ubuntu_pro_apt_news" pid=14698 comm="(python3)"
[41248.516902] audit: type=1400 audit(1726661999.698:115): apparmor="DENIED" operation="change_onexec" class="file" info="label not found" error=-2 profile="unconfined" name="ubuntu_pro_esm_cache" pid=14699 comm="(python3)"
[41906.485121] usb 1-1: USB disconnect, device number 8

https://github.com/user-attachments/assets/3d5b31a6-ce2a-4ee1-8901-1c23b57b379a

Then short B8 to B12 over a 10k resistor and plug it to the tablet USB-C connector

Connect my remarkable 2 via USB.

The tablet turns on on it's own but the screen doesn't freeze and always keep starting.

The console doesn't update or show anything as soon I connect the Pogo cable.

So I run out of ideas. Thanks for the reply btw.

Eeems, 2024-09-18:
In the logs, which lines show up when you powered on the device with everything attached? I assume you have `dmesg -w` running before you start the device, and then are watching the output?

M3TALZ3R0, 2024-09-20:
Yes, I have "sudo dmesg-w" running as the instructions say before anything else. And the tablet flashes 2 times and starts normally, dmesg shows nothing when I tried to connect the tablet via pogo.

Now I realize that the tablet doesn't blink 3 times when I lower B8 with 10k resistance when I try to spin through the power button. I have tried many different resistors and still the same results.

I was also testing different things on Windows OS and pull down B8 and then connected the tablet via pogo. To my surprise, the tablet freezes in the Power off logo, but Windows recognized the device.

M3TALZ3R0, 2024-09-20:
Okay after adding 10.10k resistor the dmesg -w detected the device>

> [ 6289.626881] usb 1-5: new high-speed USB device number 25 using xhci_hcd
> [ 6289.754345] usb 1-5: New USB device found, idVendor=15a2, idProduct=0076, bcdDevice= 0.01
> [ 6289.754362] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [ 6289.754369] usb 1-5: Product: SE Blank ULT1
> [ 6289.754374] usb 1-5: Manufacturer: Freescale SemiConductor Inc 
> [ 6289.756996] hid-generic 0003:15A2:0076.0013: hiddev3,hidraw8: USB HID v1.10 Device [Freescale SemiConductor Inc  SE Blank ULT1] on usb-0000:00:14.0-5/input0
> sudo ./imx_usb
> [ 6336.842604] usb 1-5: USB disconnect, device number 25

But the device doesn't start on it's own. So as soon I remove B8 the device disconnect itself from the laptop and then boot normally.

BagOfHolding, 2024-09-20:
I've got exactly the same problem as you at the moment - cannot get it to register that B8 has been pulled down at all - every reboot is normal, and dmesg shows nothing at all.
I've got the pogo pins soldered to the device at the moment, powers on as soon as I connect the USB to it, but never registers the breakout board.
Tested the voltage on the breakout board by plugging it into a normal USB-C port, and its pulled down to 2.5v with a 10k resistor on it, and shows ~10k resistance on the multimeter (but randomly only 3.3kish resistance if I measure it when its plugged into the RM2)

Eeems, 2024-09-20:
> I've got exactly the same problem as you at the moment - cannot get it to register that B8 has been pulled down at all - every reboot is normal, and dmesg shows nothing at all. I've got the pogo [...]

Could you open your own issue with information like your own pictures of your setup etc.

M3TALZ3R0, 2024-09-21:
> The device doesn't update the screen when booting into recovery, so it's acting as expected. It looks to me like you are trying to run imx_usb on the same console window that is currently tailing dmesg, you'll need to open a second one to run that.

So is it's acting as expected?
And no, I've open another terminal to run `sudo ./imx_usb` and apparently something is wrong because I got this error:

`$ bash: ./imx-usb No such file or directory exist.`

I have also to say that I have granted the necessary permissions to the executable and (I think) I've properly installed imx usb loader following all the instructions.

Eeems, 2024-09-21:
> So is it's acting as expected?

Correct, you are seeing the `USB HID v1.10 Device [Freescale SemiConductor Inc SE Blank ULT1]` line in your dmesg output, exactly like the instructions indicate.

> And no, I've open another terminal to run `sudo ./imx_usb` and apparently something is wrong because I got this error:
> 
> `$ bash: ./imx-usb No such file or directory exist.`

It looks like you are trying to run `imx-usb` instead of `imx_usb`, as the error doesn't match what you say you are running, and also doesn't match the file in the terminal, which is `imx_usb`.
 
> I have also to say that I have granted the necessary permissions to the executable and (I think) I've properly installed imx usb loader following all the instructions.

If you've installed imx_usb_loader, you can use that instead of the pre-compiled version in the repository. But it's probably easiest to just use the one in the repository.

M3TALZ3R0, 2024-09-21:
> Correct, you are seeing the `USB HID v1.10 Device [Freescale SemiConductor Inc SE Blank ULT1]` line in your dmesg output, exactly like the instructions indicate.

Got it

> It looks like you are trying to run `imx-usb` instead of `imx_usb`, as the error doesn't match what you say you are running, and also doesn't match the file in the terminal, which is `imx_usb`.

My bad, the actual error is:

```
sudo ./imx_usb
sudo: ./imx_usb: command not found
```

Eeems, 2024-09-21:
That would be because you aren't in the folder that contains it. Navigate to the folder you pulled down the repository to, and then run it.

M3TALZ3R0, 2024-09-22:
> That would be because you aren't in the folder that contains it. Navigate to the folder you pulled down the repository to, and then run it.

Ok, I ran it from **/bin folder** and got this:

> joel@joel-ROG-Strix-G712LV-G712LV:/bin$ sudo ./imx_usb
> config file </usr/share/imx-usb-loader/imx-loader.d//imx_usb.conf>
> vid=0x066f pid=0x3780 file_name=mx23_usb_work.conf
> vid=0x15a2 pid=0x004f file_name=mx28_usb_work.conf
> vid=0x15a2 pid=0x0052 file_name=mx50_usb_work.conf
> vid=0x15a2 pid=0x0054 file_name=mx6_usb_work.conf
> vid=0x15a2 pid=0x0061 file_name=mx6_usb_work.conf
> vid=0x15a2 pid=0x0063 file_name=mx6_usb_work.conf
> vid=0x15a2 pid=0x0071 file_name=mx6_usb_work.conf
> vid=0x15a2 pid=0x007d file_name=mx6_usb_work.conf
> vid=0x15a2 pid=0x0080 file_name=mx6ull_usb_work.conf
> vid=0x1fc9 pid=0x0128 file_name=mx6_usb_work.conf
> vid=0x15a2 pid=0x0076 file_name=mx7_usb_work.conf
> vid=0x1fc9 pid=0x0126 file_name=mx7ulp_usb_work.conf
> vid=0x15a2 pid=0x0041 file_name=mx51_usb_work.conf
> vid=0x15a2 pid=0x004e file_name=mx53_usb_work.conf
> vid=0x15a2 pid=0x006a file_name=vybrid_usb_work.conf
> vid=0x066f pid=0x37ff file_name=linux_gadget.conf
> vid=0x1b67 pid=0x4fff file_name=mx6_usb_sdp_spl.conf
> vid=0x0525 pid=0xb4a4 file_name=mx6_usb_sdp_spl.conf
> vid=0x1fc9 pid=0x012b file_name=mx8mq_usb_work.conf
> vid=0x1fc9 pid=0x0134 file_name=mx8mm_usb_work.conf
> vid=0x1fc9 pid=0x013e file_name=mx8mn_usb_work.conf
> vid=0x3016 pid=0x1001 file_name=mx8m_usb_sdp_spl.conf
> config file </usr/share/imx-usb-loader/imx-loader.d//mx7_usb_work.conf>
> parse /usr/share/imx-usb-loader/imx-loader.d//mx7_usb_work.conf
> no job found

Tried to found a solution on Google but no luck.
By the way, thanks for the help, sorry for being so dumb.

M3TALZ3R0, 2024-09-22:
Ok, apparently my tablet is having some problems, because as soon as I remove the 10k resistor, it starts up by itself and doesn't go into recovery mode.

Eeems, 2024-09-22:
Not from your bin folder, you need to run it while you are in the folder that contains the contents of this repository. Otherwise it doesn't have instructions for what to do to the device.

M3TALZ3R0, 2024-09-23:
> Not from your bin folder, you need to run it while you are in the folder that contains the contents of this repository. Otherwise it doesn't have instructions for what to do to the device.

Already tried and still the same:

```
sudo ./imx_usb
sudo: ./imx_usb: command not found
```

Eeems, 2024-09-23:
That tells me that you don't have the repository pulled down to your machine, or you aren't actually in the correct location, as `imx_usb` exists in the repository in the root folder for it.

Eeems, 2024-10-10:
@M3TALZ3R0 poke?