ddvk/remarkable2-recovery #35: Cannot get RM2 into Recovery Mode

BagOfHolding, 2024-09-21:
Opening new issue as requested

Cannot for the life of me get the RM2 into Recovery Mode.

So far I've soldered the Pogo connections to the pins to ensure that its a solid connection - these were then soldered to a USB cable, but I've since swapped that out for a shorter one that I know works (and sacrificed it) - only connection not wired on the breadboard is the ID pin from the Pogo, but I think that should be disconnected, or wired to short on one end, but I'm not sure which end (Plug end, or Pogo end)

USB-C breakout board has been tried in both orientations, with a 10k resistor hard soldered to B8 / B12

Each time the device is connected to the host machine, it immediately starts up (so I know the pins are the right way round) but then flashes twice and goes into a normal boot mode  - nothing shows on dmesg

Connecting the RM2 normally to the host machine over the usb-c port, dmesg shows it connect as the RNDIS/Ethernet Gadget as it should do under normal operation, so I know the USB-C port is ok

Measuring the ports, I've checked the breakout board on a USB extension, and B8 registers as 2.5v, VCC (b4) shows as 5v, so the breakout board is working ok, shows ~10K resistance when not connected to power, but when connected to the RM2 and under power, it shows 0v and ~3.2k resistance - not sure if this is relevant or not.

I did see on the other post that they'd changed to a 10.1k resistor, but I havent try upping that yet to see, as I'm not confident enough to know how much to increase it to.

[image]
[image]

BagOfHolding, 2024-09-21:
Read up on the OTG port requirement - connected ID to ground to short it, but still exactly the same

Eeems, 2024-09-21:
Have you tried flipping the usbc breakout board, just in case the orientation it expects doesn't match what the device expects?

Have you used a multimeter to make sure none of the Pogo connections are bridged? I know it's unrelated to getting it into the recovery mode, but it's something that I can't really tell visually from the pictures.

As for if you have the right cables for the Pogo port pins, I'll have to check my board to see which pins are passed through and which aren't.

Eeems, 2024-09-21:
https://photos.app.goo.gl/gJgn5VE4qomnpoSU9
My board uses all 5 pins to the micro usb connector. As for what the standard is for micro to full sized is, I don't remember.

f00lycooly, 2024-09-22:
Re-made the plug, using pogo connections an a USB-B breakout board, still boots normally and wont go into recovery mode.
[image]
[image]

f00lycooly, 2024-09-22:
yup - tried the USB-C board in both ways, tried adding a 1k resistor in series in case the resistance wasn't enough - remade 3 cables so far, only thing left is to get a different breakout board for the USB-C and try that, but I can imagine that changing things, as I've tried 2 separate breakout boards now

Eeems, 2024-09-22:
Okay, to confirm your steps, could you list the steps you are doing, in the order you are doing them? Also, since you've indicated that you haven't seen anything at all in dmesg, can you confirm that the cable you are using works for data transfer with another device?

f00lycooly, 2024-09-22:
Steps:

Power Off RM2 - hold power button in for at least 10 seconds - confirmed its powered off as unresponsive.
Connect USB-C Breakout board to port
Connect Host USB connection to Pogo Pins (its been done by soldered connection with 2 different cables and the new breakout board by holding it hard against the pogo pins directly, pressing to full depth, no movement off the pins)

At this point, the RM2 comes back to life - flashes twice on the Starting screen, then flashes once more as it goes into normal boot.

Nothing registers on dmesg at all - not even a bad connection message

I've ordered a brand new cable as I dont have any spare Mini USB cables (knew I shouldn't have thrown that crate of cables away....) so I'll give it another test tomorrow with that one.

Eeems, 2024-09-22:
Since you haven't confirmed that the cable you are using supports sending data with another device, it does kind of sound like it only supports power, which is why you wouldn't get anything in dmesg.

If you wait long enough in recovery mode without sending a payload, it will automatically reboot into the normal boot process. Which is what it sounds like is happening with your device.

Eeems, 2025-07-21:
@f00lycooly did you ever get this working?

tgalal, 2025-08-07:
Stumbled upon same issue, turned out the pin labels on my board are printed wrong. They are inverted horizontally and I think vertically as well. This seems like a common problem in those breakout boards, see my real B8 in the attached image which worked flawlessly. Before noticing, it was the exact symptoms like OP.

Looking at OP's pictures, horizontal might be okay, perhaps shorting A5 instread of B8 would get it working.

[image]