ddvk/remarkable2-recovery #38: Error When Running grep Developer Command

Xelmarik, 2024-12-30:
Hello,

I've been able to place my Remarkable 2 in recovery. when I do so, it auto-mounts my partitions, these are as follows:
1. /dev/sdb1 (Boot, mounted)
2. /dev/sdb2 (283 Mb partition, mounted)
3. /dev/sdb3 (283 Mb partition, unmountable, I assume this was corrupted when I bricked the device)
4. /dev/sdb4 (Home, mounted)

I assume sdb2 is my fall back partition created when I downgraded my Remarkable 2, and that sdb3 was the one I was booting to when I failed to remove all the mods before factory reset. I have the original SSH password that I had been using, so I'm not sure if I need to try to recover that.

But when I run the `grep Developer /dev/sdb4/root/.config/remarkable/xochitl.conf` I get the error `grep: /dev/sdb4/root/.config/remarkable/xochitl.conf: Not a directory`

When I go into my file folder I can see that there is no .config in the root folder. Not sure where it went. 

My question is what do I need to do next then? Should I try to SSH and then run a "dd" command to go back to the other partition? If I can't SSH, do I need to rebuild the config? Just not sure since the grep Developer doesn't return the SSH password.

Thank you.

Eeems, 2024-12-30:
Are you able to SSH in still?

Xelmarik, 2024-12-30:
Just tried and it doesn't give me the option to enter my password, then I get a SSH time out error.

Eeems, 2024-12-30:
Does the UI work?

Xelmarik, 2024-12-30:
No, it's in recovery mode displaying the start up screen.

Xelmarik, 2024-12-30:
I should note, I'm connect via the pogo pin connector right now, not a USB-C cable like I normally would for SSH.

Eeems, 2024-12-30:
Recovery does not support SSH, and SSH only works over wifi and usb-c. You only get access to the block devices.

I assume the device is not working normally right now?

Xelmarik, 2024-12-30:
Got it. Yeah, the device does not work at all, it's soft bricked right now.

Eeems, 2024-12-30:
Alright, just replace both root partitions with the raw image from an upgrade file. See https://github.com/ddvk/remarkable2-recovery/pull/27 for in-progress instructions.

Xelmarik, 2024-12-30:
Thanks, I'll try to follow that and if I have any questions, I'll let you know.

Xelmarik, 2024-12-31:
To close this issue out, the instructions in #27 worked wonderfully. Below is the extract of my terminal running the commands (included a when I failed to edit certain parts of the back ups and me correcting that). My Remarkable 2 is now fully restored and working. Thank you all for your support and I hope to support you going forward for your great efforts!

`[xelmarik@flashendeavor ~]$ sudo dd if=/dev/sdb2 of=rm2-sdc2.raw bs=4096 status=progress
[sudo] password for xelmarik: 
282054656 bytes (282 MB, 269 MiB) copied, 24 s, 11.8 MB/s
69120+0 records in
69120+0 records out
283115520 bytes (283 MB, 270 MiB) copied, 24.1422 s, 11.7 MB/s
[xelmarik@flashendeavor ~]$ sudo dd if=/dev/sdb3 of=rm2-sdc3.raw bs=4096 status=progress
282710016 bytes (283 MB, 270 MiB) copied, 24 s, 11.8 MB/s
69120+0 records in
69120+0 records out
283115520 bytes (283 MB, 270 MiB) copied, 24.0889 s, 11.8 MB/s
[xelmarik@flashendeavor ~]$ sudo dd if=rm2-sdb2.raw of=/dev/sdb3 bs=4096 status=progress
dd: failed to open 'rm2-sdb2.raw': No such file or directory
[xelmarik@flashendeavor ~]$ sudo dd if=/dev/sdb2 of=rm2-sdb2.raw bs=4096 status=progress
281530368 bytes (282 MB, 268 MiB) copied, 24 s, 11.7 MB/s
69120+0 records in
69120+0 records out
283115520 bytes (283 MB, 270 MiB) copied, 24.185 s, 11.7 MB/s
[xelmarik@flashendeavor ~]$ sudo dd if=/dev/sdb3 of=rm2-sdb3.raw bs=4096 status=progress
281006080 bytes (281 MB, 268 MiB) copied, 24 s, 11.7 MB/s
69120+0 records in
69120+0 records out
283115520 bytes (283 MB, 270 MiB) copied, 24.2416 s, 11.7 MB/s
[xelmarik@flashendeavor ~]$ sudo dd if=rm2-sdb2.raw of=/dev/sdb3 bs=4096 status=progress
69120+0 records in
69120+0 records out
283115520 bytes (283 MB, 270 MiB) copied, 28.278 s, 10.0 MB/s`

Thank you especially to @Eeems for your quick replies and instructions.

closed 2024-12-31 by Xelmarik