ddvk/remarkable2-recovery #39: Is it possible to use a USB 3 or USB C breakout instead of micro-usb

phaer, 2025-01-16:
Thank you for this awesome project, I am trying to recover my remarkable 2 after I soft-bricked it. 

I don't have a micro usb breakout board at hand, but do have the rest of the parts including an 8 pin and a 24 pin usb c breakout board.

The thing is: they don't feature the USB OTG "id" pin, at least not under this name. And I don't know whether that pin is relevant for the recovery to work or whether it can be replaced.

So is it sufficient to have ground, sbus, d+ and d- for the recovery? Or is there a trick to use a 24 pin usb c breakout board?

phaer, 2025-02-11:
I am reading this as saying the ID pin might not be needed? Was a bit hesitant to just try it out, but I guess the risk of frying my device here is pretty low so going to try it this weekend or so and let you know. Thanks!

ddvk, 2025-02-11:
iirc, the id pin is not used in the recovery, the device is slave

cret0103, 2025-07-27:
Thank you for sharing the method！ 
Can I directly use the usb to pogo adapter in the following drawing?
[image]

HelloDaiAn, 2026-09-05:
@cret13 @N4SIRODDIN3 Hi~ does this excellent idea work?