ddvk/remarkable2-recovery #41: Device not mounting Drives after reconnection

MRUnbekanntOP, 2025-02-10:
I followed the guide and got to the point where i successfully mounted the partitions of my RM2. I, then, tried to unbrick it, but wasn't successful. after that, i disconnected it from my notebook and tried booting it, but it didn't show any changes on the display. After that, i tried to connect it again via the pogo pins, which resulted in the following dmesg message: 
`[ 5600.164463] usb 1-8: new high-speed USB device number 32 using xhci_hcd
[ 5600.292016] usb 1-8: New USB device found, idVendor=0525, idProduct=a4a5, bcdDevice= 2.21
[ 5600.292035] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5600.292040] usb 1-8: Product: USB download gadget
[ 5600.292045] usb 1-8: Manufacturer: FSL
[ 5600.292048] usb 1-8: SerialNumber: 000000a7ec689209
[ 5600.293658] usb-storage 1-8:1.0: USB Mass Storage device detected
[ 5600.293875] usb-storage 1-8:1.0: Quirks match for vid 0525 pid a4a5: 10000
[ 5600.293937] scsi host4: usb-storage 1-8:1.0
[ 5601.428137] usb 1-8: reset high-speed USB device number 32 using xhci_hcd
[ 5601.676672] usb 1-8: reset high-speed USB device number 32 using xhci_hcd
[ 5601.925783] usb 1-8: reset high-speed USB device number 32 using xhci_hcd
[ 5602.175161] usb 1-8: reset high-speed USB device number 32 using xhci_hcd
`
from there, i seemed to be unable to remount the partition of the device. also it doesn't react at all to any attempts to reboot it back into recovery mode. when I look at the lsusb output, the device is visible as 
`Bus 001 Device 032: ID 0525:a4a5 Netchip Technology, Inc. Linux-USB File-backed Storage Gadget`
but it is not visible in the output of lsblk. does anyone have an idea on how to gain access to the drives again?

Eeems, 2025-02-10:
What steps did you take to "unbrick it"?

MRUnbekanntOP, 2025-02-10:
I sadly do not remember everything, but i do know that i tried copying the xochitl.service-file between the two boot partitions, and i tried completly flashing boot partition sdb2. both of them did not work. I also wanted to mention that i had this problem before, and i was able to get around it by reentering recovery mode, which no longer works.

Eeems, 2025-02-10:
What did you flash the boot partition with? This likely is the cause.

MRUnbekanntOP, 2025-02-10:
I used codexctl to download and extract the firmware, then flashed it using dd. is there any way to regain access to the device if that was the problem?

Eeems, 2025-02-10:
Okay, so you replaced the second partition with the latest OS update. Which is not the boot partition, it's a root partition. You should still be able to get into recovery mode.

@ddvk any ideas why it would be giving a different device identifier than we expect?

ddvk, 2025-02-10:
no idea, you really sure you didnt overwrite the boot partition by mistake?

MRUnbekanntOP, 2025-02-10:
Yes, as far as i remember there were 4 partitions that were Mounted. One was called Home (sdb4), one was called boot (sdb1) and two Had seemingly random number and letter combinations as names (sdb2 and 3). I flashed the new contents on sdb2, one of the random number partitions.