ddvk/remarkable2-recovery #42: not able to see mount, running on pi4/deb6.6

ramasdf-ai, 2025-02-12:
howdy

i am trying to get my device unlocked,

i am running on `Linux 6.6.62+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux`

i see this in dmesg `[337759.282055] usb 1-2: USB disconnect, device number 42
[337760.522904] usb 1-2: new high-speed USB device number 43 using xhci-hcd
[337760.672969] usb 1-2: New USB device found, idVendor=15a2, idProduct=0076, bcdDevice= 0.01
[337760.672974] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[337760.672977] usb 1-2: Product: SE Blank ULT1
[337760.672980] usb 1-2: Manufacturer: Freescale SemiConductor Inc
[337760.675722] hid-generic 0003:15A2:0076.0028: hiddev97,hidraw2: USB HID v1.10 Device [Freescale SemiConductor Inc  SE Blank ULT1] on usb-xhci-hcd.0-2/input0`

this usb device disconnects and reconnects every 3 to 4 minutes. 

running imx_usb, give be the error below.
`$ sudo ./imx_usb
./imx_usb: 1: Syntax error: ")" unexpected`

i tried added to udev and reloaded, no change. 

also this step is unclear, does b12 need to stay in the whole time? and only b8 is being pulled down and then pulled up? 

`pull down B8 (connect B8 over a 10K resistor to B12)`

will try a new cable, after this. 

- Rama

ddvk, 2025-02-14:
you need usb-c as he B8
you don't need microusb at all, just something to connect to the pogo pins.

imx-usb is a x64 build it won't work on aarch64, unless you build it yourself.

ramasdf-ai, 2025-02-14:
well, that pretty confusing, **the parts and images the repo has are micro usb.**

if i get a usb c, then the [holes have](https://mm.digikey.com/Volume0/opasdata/d220001/medias/images/309/BOB-15100.jpg) different digits on them.  then it turns in which pogo etc. 

is it possible for y'all to update the parts list what really needed and not outdated microusb?

Eeems, 2025-02-14:
Looking at the parts list and the images, while it's using the digikey pre-filled cart that is in the parts list (not the aliexpress parts), they both do still have a micro usb breakout for the pogo port, and a usb-c breakout for the usb-c port.

ddvk, 2025-02-15:
the point of confusion is the usb-c cable probably, which was meant as a 'usb-c to micro usb'.
someone asked whether usb-c would work as well, and it would if you would connect the correct pins.

the setup with the microusb is the easiest, tested and working version.

ramasdf-ai, 2025-02-17:
well, i am probably at my last hope, i bought a 4 port usb c dongle to power the pi and plug usb-c to micro usb pogo. tried running this stuff from a mac and that didn't work out well either. maybe will try posting photos before i do a hard reset. this is a major pain and a big loss if i need to hard reset.

ramasdf-ai, 2025-02-19:
So coming back and just want to thanks that i got my remarkable unlocked.  

I was pretty frustrated about not remembering my passcode and it took me a ways to figure out how to do all the steps listed here.  Instead of trying compile on arch, i ended up installing refind and endeavouros on my mac book (x86_64), to which i was able to use short usb-c to microusb cord. 

I would like to contribute back by updating steps/readme. how do i go about that?

Thanks again! really saved a good deal of pain.

closed 2025-03-01 by ramasdf-ai

machuwey, 2025-04-05:
> So coming back and just want to thanks that i got my remarkable unlocked.
>
> I was pretty frustrated about not remembering my passcode and it took me a ways to figure out how to do all the steps [...]

Hey man, I have the same issue, could we please talk `matveyder@protonmail.com`. I'm willing to pay

machuwey, 2025-04-07:
Hey man @ramasdf-ai

Eeems, 2025-04-07:
@machuwey Please do not spam a closed issue asking someone to email you. If you need help you can open your own issue, or get help on the community discord.