ddvk/remarkable2-recovery #43: What do you do after successful mount

denull0, 2025-02-18:
Basically I have few partitions 
1 - boot
2 - can't mount 
3 - can't mount 
4 - mounts its around 7GB and has a root folder

In the boot file I tried setting the active partition to all but I am still stuck on the starting screen. 

Any help is appreciated 🙏

ramasdf-ai, 2025-02-19:
try checking root folder.  you might need to run `ls -al` to see all the hidden files. once you see things under `/root/.config/`, follow the path listed to find the config file that has info you are looking for. 

starting screen is going to stay lit until usb mount is disconnected or device is restarted via power button.

denull0, 2025-02-19:
And then? Is there a way to factory reset? I think I set the wrong boot partition or something via ssh before I bricked it. Any idea how to revert that?

1Gitaround, 2025-02-26:
ls -al ~/.ssh

closed 2025-07-21 by Eeems