ddvk/remarkable2-recovery #45: No able to boot into recovery

machuwey, 2025-04-16:
Followed all the steps, I'm aware I'm not using linux I'm connecting to my mac host to test it out and get it onto recovery mode.

As you can see in the video I'm not able to get it into recovery. After flashing it proceeds into normal boot flow.
Was wondering what I could be doing wrong. 
Cheers.

Eeems, 2025-04-16:
This would indicate that your usb-c breakout board and 10k resistor isn't properly pulling down the pin.

machuwey, 2025-04-21:
> This would indicate that your usb-c breakout board and 10k resistor isn't properly pulling down the pin.

Hi @Eeems ,

As you correctly pointed out, the issue was indeed with the USB-C breakout board. After switching to the correct breakout board (RIGHT in the image), I was able to successfully trigger recovery mode.

[image]

However, I am concerned that breakout boards similar to the one on the left in the image might not be properly supported or could potentially be faulty. I've noticed another user encountering similar issues with the same board type (#35). It might be worth investigating or mentioning in the documentation to avoid confusion in the future.

P.S.: Could you please remove my comments from [issue #42](https://github.com/ddvk/remarkable2-recovery/issues/42#issuecomment-2780701058)? One of them contains personal information. My apologies for any inconvenience or unintended spamming.

closed 2025-04-21 by Eeems