ddvk/remarkable2-recovery #47: After typing sudo ./imx_usb i got "command not found" error.

czullo, 2025-06-12:
I dont know what im doing wrong, my remarkable is detected but when i type ./imx_usb i got command not found. Anybody can help me with that?

[image]

czullo, 2025-06-12:
all steps before sudo ./imx_usb was succesfull dont know if im doing right with imx_usb_loader i just downloaded https://github.com/boundarydevices/imx_usb_loader to desktop and unzipped it and cd imx_usb but after that i got command not found, dont know if i must compile it or something im not into linux.

czullo, 2025-06-12:
Then I remove the resistor, the device turns on automatically, then I go to https://github. com/ddvk/remarkable2-recovery, I clicked the <> Code > Download ZiP button, unzipped the archive on the Linux desktop, entered the unzipped directory, and after typing sudo ./imx_usb, I get the message shown in the picture below.

[image]

Eeems, 2025-06-12:
What does `uname -a` on your computer return?

czullo, 2025-06-13:
[image]

Eeems, 2025-06-13:
What happens if you just try to run `./imx_usb` from the `~/Desktop/1` directory?

czullo, 2025-06-13:
I tried another Linux distribution, Ubuntu 25.04, and encountered the same error as on the Kali distribution.

[image]

Eeems, 2025-06-13:
You remove the 10k resistor before you run imx_usb.

czullo, 2025-06-13:
After removing the resistor, remarkable2 starts up normally, and after entering the command sudo ./imx_usb, it displays “No machiung USB device find” as in the previous photo.

czullo, 2025-06-13:
Finally, it worked. Now, after entering the command:

grep Developer sdd4/root/.config/remarkable/xochitl.conf

I get the message:

No such file or directory

Eeems, 2025-06-13:
What is sdd4?

Eeems, 2025-06-13:
/dev/sdd4 is the block device, the second path is where it is mounted.

czullo, 2025-06-13:
Did I write it down correctly now?

[image]

Eeems, 2025-06-13:
https://support.remarkable.com/s/article/Passcode

czullo, 2025-06-13:
I managed to find the passcode lower down in the file; I just had to scroll down, and the device unlocked. Thank you again for your patience and tremendous help. Thank you!

[image]

closed 2025-06-13 by czullo